prusa3d/PrusaSlicer #2156: Brim does not stick to object with elephant foot compensation

espr14, 2019-04-22:
1.42.0 b2 win64

Brim should stick to the object to hold it on the bed. Reducing size of the first layer by elephant foot compensation causes brim not to stick to the object. It does not stick in some parts only. These non-stick parts could be quite long. It is not symmetrical for symmetrical object so there's something very weird.

[image]

x0rtrunks, 2019-04-22:
If this does get fixed, can I request this to be an option feature? Call it easier to remove brims or something.

espr14, 2019-04-23:
That's good idea. Still, it is better to define distance of brim from object. Leaving gaps could result into the same like without brim due to bending.

supermerill, 2019-04-23:
If your elephant foot setting is correctly tuned, your brim will connect with the piece irl, as the objective is to "repair" the over-extruding first layer and not to create a chamfer.

espr14, 2019-04-23:
That makes sense but in this case there are parts where brim is connected and parts disconnected. That doesn't make sense.

supermerill, 2019-04-23:
oh, right, i didn't see that. weird. could you post your platter as a .3mf (save project & zip it) ?

espr14, 2019-04-23:
Here's 0.2 compensation which is even worse. You can try 0.3 to get upper image. Btw. 3mf is already a zip so renaming is sufficient.

[elep.3mf.zip](https://github.com/prusa3d/Slic3r/files/3107278/elep.3mf.zip)

bubnikv, 2020-01-02:
The elephant foot compensation improvements are part of PrusaSlicer 2.2.0-alpha2. Closing.

closed 2020-01-02 by bubnikv