prusa3d/PrusaSlicer #2158: The French version does not work, and it is impossible to change the speed, modify in the tab it does not impact the whole program as if it were bridling somewhere.

CHATON89, 2019-04-22:
### Version
_Version of Slic3r Prusa Edition used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

bubnikv, 2019-04-23:
sorry, but you did not provide any information on the Slic3r version,
also "The French version does not work" is a little too generic.

BastR, 2019-04-24:
This also happens to me on macOs and I have seen it on Windows 64bits as well.

In version 1.42.0 beta2+, the French translation does not work, the software remains in English.

BastR, 2019-04-24:
Ok a rapid update, I've downloaded the previous version (1.42.0 beta 1) and put "Resources/locallization/fr_FR/**Slic3rPE.mo**" from the beta 1 to the same folder in the beta 2 and it work.

I think "Slic3rPE.mo" is the same in the English folder and the French folder the beta 2

bubnikv, 2019-04-24:
we are currently working on translating the current Slic3r PE.

BastR, 2019-04-24:
Can we help?

CHATON89, 2019-04-24:
I turn on 1.42 beta 2, For the translation I did the same I recovered the file on my alpha version 7 because otherwise the software remains in English, For the speed problem I had to disable the cooling function automatic otherwise the speed remains constrained by the value of the minimum speed parameter.

YuSanka, 2019-04-25:
Thanks for your report. It looks like mysterious )) But, yes French file was changed to the English :( Fixed PO and MO will be updated in master.

bubnikv, 2019-05-04:
The language issue will be solved with the next release.

> For the speed problem I had to disable the cooling function automatic otherwise the speed remains constrained by the value of the minimum speed parameter.

Please don't mix multiple issues in a single report. If it bothers you, please open another issue and provide as much inormation as possible.

closed 2019-05-04 by bubnikv