prusa3d/PrusaSlicer #2159: Rename the app file to "Slic3r PE.app" (macOS)

DominikPalo, 2019-04-23:
It would be nice if the Prusa Edition of Slic3r has a different name of the app file (packed in DMG) than the original Slic3r. Now both slicers share the same name "Slic3r.app" and that causes problems for users (like me) who use both Slic3rs. It happened a few times that I accidentally replaced the original Slic3r in my Applications folder with a Prusa Edition while I was trying to update the Prusa Edition. 
Now (during every PE update) I have to copy the app file from DMG outside, rename it manually to "Slic3r PE.app" and then copy it to the Applications folder. So, I suggest using the "Slic3r PE.app" name for the Prusa Edition to prevent such conflicts.

x0rtrunks, 2019-04-24:
Yes please.

bubnikv, 2019-05-04:
it will be renamed with the next release.

closed 2019-05-04 by bubnikv