prusa3d/PrusaSlicer #2160: Do not slowdown on external perimeters

guestisp, 2019-04-23:
In the cooling page, there is an option to slow down printing speed up to a minum layer print time.
When slowdown happens on external perimeters, certain bowden extruders could make some blobs or zits on the visibile surface.

Slow down should only happen, as much as possible, in the infill or in the internal perimeters, so that any blobs would be hidden.

If you print with infill first, the perimeters are always printed after the infill, thus slowing down during the infill allow the previous layer perimeters to coold down
If you print in standard way, with perimeters first, the next infill will slow down allowing the previous perimeter to slow down anyway.

I don't see any drawbacks in slowing down only infill or the internal periemters.

bubnikv, 2019-05-04:
I am sorry nothing will really fix your bowden extruder problem to a full extent. We have that feature you described, and we had to rework it, as it lead to extreme slow downs over the infill before slowing down the perimeters, leading to gaps at the perimeters following the extremely slowed down infill.

guestisp, 2019-05-04:
I understand your point of view, just as side note, it's not only a bowden issue, i've printed the same gcode even with a a direct extruder on a different printer. I'm trying to find an origianl prusa to see if the same broken gcode (sliced by Slic3r PE) will lead to the same bad extrusion even on a Original Prusa.

It's a seam issue, whenever there is a seam, there are tons of holes and lines, i've also seen multiple reports about the same on reprap forum and prusa forum. In example, I'm able to print perfectly a vase, with no issue at all as there isn't any seam in a vase.

(I can see the same issue even on a multi-loop skirt, during the loop change)

VanessaE, 2019-11-21:
(using commit a1aa343 here)

This is definitely a problem, and you are missing a key problem that's independent of the hardware:

Varying the speed of external perimeters affects their shine and quality (especially with things like PETG, which likes to take it slow already), adds extra heat to the perimeter, and introduces minor thickness differences from layer to layer, probably owing to the filament's die swell or its shrink ratio.  These artifacts combine to make the outside of the part look less smooth than the printer is capable of.

Upstream Slic3r varies the speed of infill, interior perimeters, and gap fill to manage layer time, while keeping the external perimeters at their normal speed, whether you're using configured speeds or autospeed/volumetric mode.  Or, it's *supposed* to anyway, because of the above issues (though lately I've noticed that it sometimes slows down exterior perimeters too, with no obvious reason, which is why I decided to give PrusaSlic3r another go).

@bubnikv  I also realize that large differences in flow rates between temporally-adjacent features can cause blobs or starvation at their starting points, but that's not an unsolvable problem, and as @guestisp said, it is not limited to bowden extruders.  You could fix it with a variation of Slic3r's existing autospeed/volumetric behavior.  For example:

Assume the slow-down feature is invoked for a given layer, and that it has decided that infill, interior perimeters, and gap fill should all print super slow relative to normal, while the exterior perimeter prints at its normal speed (thus relatively fast).  I'll also assume the most common feature print order (just to simplify my description):  interior perimeters from innermost to outermost, then the exterior perimeter, then infill and/or gap fill.

Split the last 10 mm or so of the final interior perimeter into a few dozen segments, then simply ramp the speed up over that interval, so that the flow rate at the end of that perimeter matches that of the exterior perimeter (at its full speed).  Presto, no perimeter starvation, and no complicated algorithms :smile:.  And yes, you'd do this even if a retraction/travel between the two is indicated, as pressure in the nozzle matters even when punctuated with retractions.

When you move from the exterior perimeter to infill, or from infill or gap fill to the interior perimeter of the next island/layer, you could also ramp up/down, but this would be pointless -- who cares if sparse infill or the innermost perimeter have mild blobs or starvation if they're hidden?

The ramping distance would have to shrink as islands get smaller, and at some point, some island would be too small for most of it to print at the slowest speeds, but I guess at that point you'd just jump up and down in speed like normal, not bother ramping up/down.

Naturally, you wouldn't bother to take these speed changes into account when computing layer time and target speeds.  You'd just apply some "fudge" instead (maybe just further slow down the already-slower stuff by another 5% or something), as no one's gonna care if a layer is a few seconds slower than the "ideal" time.

As for seams, that's a different issue with its own solutions which could be applied independently of proper flow rate management.

closed 2026-09-29 by prusaslicer-bot[bot]