prusa3d/PrusaSlicer #2161: Beta-2: Strange slicing behaviour

Boby71, 2019-04-23:
### Version
1.0.42-Beta2

### Operating system type + version
Win10/64Bit

### 3D printer brand / version + firmware version (if known)
i3 MK2.5s Prusa-Firmware 3.7.0 based on Marlin FIRMWARE

### Behavior
A solid model get's a very strange slicing on the surface (see screenshot)
[image]
Both handrails of this bridge are the same in the cad drawing; nr. 2 is a copy of nr. 1!
There is also no visible problem in the 3D preview:
[image]

_Is this a new feature request?_
Nope, looks like a bug

#### Project File (.3MF) where problem occurs
[Fahrersteg v9.zip](https://github.com/prusa3d/Slic3r/files/3108926/Fahrersteg.v9.zip)

supermerill, 2019-04-25:
[image]

your .3mf is bad, it's not a bug.

bubnikv, 2019-05-04:
One of the hand rails is slightly tilted towards the print bed plane. Orienting it to the print bed using the "Place on face" tool fixes your issue.

closed 2019-05-04 by bubnikv