prusa3d/PrusaSlicer #2162: Random aligned seams

espr14, 2019-04-23:
1.42.0 b2

Aligned seams are great but when they appear just above each other it can cause gaps. Typically, simple models have some similar sharp corners. So the algorithm could find the sharpest corner and then all the others that have angle up to e.g. 10% bigger. From this set it would pick one randomly.

For example this square has 4 sharp corners but only one is used for seams. With the improved algorithm it would use all the 4 corners.

[image]

guestisp, 2019-04-24:
So, they wont be aligned........

espr14, 2019-04-24:
Yes. In fact there should be some changes: 
To be renamed:
- nearest -> nearest hidden
- aligned -> aligned hidden
To stay;
- random (random position, non-overhang if possible)
New ones:
- random hidden (proposed one)
- nearest (nearest position, non-overhang if possible), great for drafts
- aligned (does not make much sense)

guestisp, 2019-04-30:
Same issue here with this retraction test: https://www.thingiverse.com/thing:909901
When the tower is small (more or less, after Z 13 or 14), the seam is placed every time in the same corner, making a very rough and bumpy surface

haarp, 2019-07-21:
A "random hidden" seam option would be great. The "hiding" algorithm is quite powerful and useful.

But I strongly think that "Nearest" and "Aligned" in general are very undescriptive. Unless you study the original Slic3r manual (find it [here](https://manual.slic3r.org/expert-mode/fighting-ooze)), you wouldn't think that they would try to hide anything. So ironically, the fact that PrusaSlicer is hiding seams, is very hidden.

My proposal is not to introduce more seam options, but to introduce a checkbox "Try to hide seams".

Nearest + "Try to hide" disabled: Just put them wherever it's nearest
Nearest + "Try to hide" enabled: Find hidden spaces, but within those, put them wherever it's nearest
Random + TtH disabled: True random
Random + TtH enabled: Find hidden spaces, but within those, distribute them randomly
etc.

Felix-fh, 2020-05-19:
honestly I don't see how this seam on this single perimeter part is "random"
it seems to be just shifted some degrees. that way I don't get the robust structure I was hoping for
[image]

smartroad, 2020-09-15:
I was thinking about this, could the modifier blocks be used to specify an area where retraction can take place? Could use a shape to specify an area where seams can be, say at the rear of the model where it wouldn't be seen. Because it would be an area the existing, nearest, aligned and random can still be used, just only in the area specified by the modifier block.

neophyl, 2020-09-15:
In your case just set the seam location to rear and position the model so that the back of it where you want the seam is at the rear of the bed.

smartroad, 2020-09-15:
> 
> 
> In your case just set the seam location to rear and position the model so that the back of it where you want the seam is at the rear of the bed.

I am just using my example as a, well, an example lol. To be honest my suggestion would be more ideal with the "random" option as it prevents a seam which could be a weaker point but still allows control over where the blobs might appear. The rear option still creates a seam.

neophyl, 2020-09-15:
Every option creates a seam, its just where to put it.  Hiding it in corners is usually the best option if there are corners available.

smartroad, 2020-09-15:
Random doesn't create a seam, which I believe is the point of it. The downside is it is truly random all over the piece which can lead to odd markings over the whole surface.

neophyl, 2020-09-15:
I think we have a problem with the definitions we are using being different.  To me a seam is a seam, regardless of if its all in a line, hidden in the corners or spread over the entire object in different positions.  Its still a seam, just not all in one spot.  My definition is the start end of a layer where the perimeters meet is the seam.

Thats away from the original feature request anyway.  Has anyone else used the new seam placement tool in the last version of Super Slicer ?  Its a bit rough around the edges but it allows you to place one or more sphere modifiers and the seam for each layer is placed as close as it can be to the nearest seam modifier object.  That would 'seem' to do roughly what you suggest with the modifier mesh albeit implemented in a different fashion. As Super Slicer is a fork of Prusa Slicer then if its possible in one it should be possible in the other :)

lukasmatena, 2020-09-15:
@smartroad  @neophyl 
We are currently working on custom seam enforcers/blockers for PrusaSlicer 2.3. It will allow to paint enforcing and blocking areas on your model, similarly to what you have just described. So in @smartroad 's example, he would just paint the front by a blocker to exclude that area from potential seam locations (or he could paint enforcer at the back as he suggested to get similar outcome).

While we are at it, we should also fix the not exactly random seam that @Felix-fh posted. This is not in `master` yet.

bubnikv, 2020-10-17:
> While we are at it, we should also fix the not exactly random seam that @Felix-fh posted. This is not in master yet.

This should be fixed in PrusaSlicer 2.3.0-alpha1. Closing.

closed 2020-10-17 by bubnikv