prusa3d/PrusaSlicer #2163: Unwanted Perimeter overlap

hevilp, 2019-04-24:
### Version
Slic3rPE-1.42.0-beta2+win64-full-201904140830

### Operating system type + version
Win10 x64

### 3D printer brand / version + firmware version (if known)
Voron V2.1

### Behavior
Slicer PE seems to make a perimeter overlap, which is hardcoded and cannot set in the options.

I sliced a 20mm calicube with forced extrusion width 0.4 every field and 2 perimeter.

The gcode which comes out first lines:

```
G92 E0 ; reset extrusion distance
G1 X109.443 Y109.443 F7800.000 ; move to first perimeter point
G1 E2.00000 F2400.00000 ; unretract
G1 F1800
G1 X90.557 Y109.443 E2.19081 ; perimeter
G1 X90.557 Y90.557 E2.38162 ; perimeter
G1 X109.443 Y90.557 E2.57243 ; perimeter
G1 X109.443 Y109.368 E2.76248 ; perimeter
G1 X109.800 Y109.800 F7800.000 ; move to first perimeter point
G1 F1800
```

I take the lines with `move to first perimeter point`
The first perimeter is X109.443 Y109.443 
the second is X109.800 Y109.800

difference from each axis is 0.357 mm. Why?
It should be 0.4, so that each line is exact 0.4 width, like is set it in the option.

That means, that the width of the 2 perimeter are not 0.8mm, like I want it and set it in the option.

Why does Slic3r PE makes an (unwanted) overlap to the perimeter?

[20mm_cube_open.zip](https://github.com/prusa3d/Slic3r/files/3111369/20mm_cube_open.zip)

[20mm_cube_open.amf.zip.amf.zip](https://github.com/prusa3d/Slic3r/files/3111373/20mm_cube_open.amf.zip.amf.zip)

bubnikv, 2019-04-24:
Does this document explain what you see?
https://manual.slic3r.org/advanced/flow-math

hevilp, 2019-04-24:
There is the explanation, but why there is no option field to set the factor?
Does it need to be hardcoded?

bubnikv, 2019-04-24:
The factor is not hard coded, it is calculated from the presumed extrudate
geometry to fill in the space between two extrudates completely with the
assumption that the extrudate has rounded sides if not touching another
extrudate, and that the extrusion width is given for the rounded extrudates.

bubnikv, 2019-04-24:
> Okay,
>
> how do I get my part sliced to 0.8 skin thickness with 2 perimeter?
> And how do I get this in general, if I do 5 perimeter I'm loosing 0,215 mm
> (0.2 LH, 0.4 width), which can be quite alot.

This is another evergreen. See the following hint:

[image: image.png]

hevilp, 2019-04-24:
no pic there :/

bubnikv, 2019-04-24:
[image]

hevilp, 2019-04-24:
Okay thanks,

anyway would be nice to have an option to deactivate the behavior. It is a real nice feature, but maybe sometimes unwanted.

lixxbox, 2019-04-24:
Hi, I also noticed that recently.
Would it be possible to add an option to disable the spacing calculation? So that spacing = extrusion_width instead of spacing = extrusion_width - layer_height * (1 - PI/4).

Then it would behave more like Simplify3d or Cura, which would make it easier for me (and others) to take the values between slicers.

bubnikv, 2019-04-24:
>  anyway would be nice to have an option to deactivate the behavior. It is
a real nice feature, but maybe sometimes unwanted.

What do you propose?

> Would it be possible to add an option to disable the spacing calculation?
So that spacing = extrusion_width instead of spacing = extrusion_width -
layer_height * (1 - PI/4).

> Then it would behave more like Simplify3d or Cura, which would make it
easier for me (and others) to take the values between slicers.

First it would not be so simple to make it configurable, second it would
make the outer dimensions of the print imprecise (wider). Also having
another configuration option would make the application more confusing, and
it would put more pressure on us to track who had which option enabled.

th33xitus, 2019-04-24:
That's something i came across a few days ago while i wanted to print something.
What "annoyed" me was the fact that i wasn't able to use my forced EW together with a forced LH since the object then had features which would not be printable (result would have been holes in the perimeters). The reason for that was the flow math @lixxbox mentioned. 

I would really like to be able to "disconnect" the spacing from the layerheight. A simple checkbox would be sufficient enough. And if it is checked replace the current equation for spacing with `spacing = extrusion_width `

bubnikv, 2019-04-24:
>  I would really like to be able to "disconnect" the spacing from the
layerheight. A simple checkbox would be sufficient enough. And if it is
checked replace the current equation for spacing with spacing =
extrusion_width

But then you would have the object printed wider than it should be.

hevilp, 2019-04-24:
As I know slic3r pe object are smaller than the stl model, because you go from infill to outer perimeter, isn't it?

bubnikv, 2019-04-24:
>  As I know slic3r pe object are smaller than the stl model, because you
go from infill to outer perimeter, isn't it?

I don't understand. Slic3r calculates the extrusions, so that the
extrusions touch the target object wall.

Stephan3, 2019-04-24:
You can design a hollowed cylinder. 20mm outer diameter, inner diameter should be 18mm(5 walls .4 extrusion width). resultin in an inner diameter of 18,2mm. infill ?will? compensate this.

lukasmatena, 2019-04-24:
I probably don't understand what problem are you trying to solve. Setting spacing equal to the required layer height will always result in unwanted void space in the print, because there is no way for the printer to make the extrusion rectangular, it will always be rounded. In other words, if you print a solid cube 20x20x20 mm, you should extrude 8000 cubic mm of plastic. And in order to do that, you must make an estimate of the actual cross-section of the extruded plastic and compensate by placing the adjacent lines closer than their set extrusion width, because the extrusion width is the width we aim for at the widest place. The correction depends on layer height, as the document linked above explains. I don't see a way to go past it nor a reason to try.

Slic3r aims at printing the object as precisely as possible (with respect to the model dimension). By setting extrusion width you already fixed the offset from the model contour to the center of outer perimeters. The math above is used to determine extrusion speed to have the wall where it should be. Internal perimeters are then spaced so that proper volume of plastic is extruded and no voids left behind. It was not the goal of Slic3r to aim at specific number of perimeters - we have gap fill to take care of whatever is left inside (although I admit it has its limits). If you want specific number of perimeters, you should lower the extrusion width (but consider the diameter of your nozzle).

There was a similar issue some time ago: https://github.com/prusa3d/Slic3r/issues/1405

hevilp, 2019-04-24:
Slice this and tell me the diameters of sliced objekt (real in gcode, not in the slicer window)
Then messeaure the STL

[Quick_thread_test.zip](https://github.com/prusa3d/Slic3r/files/3112375/Quick_thread_test.zip)

lixxbox, 2019-04-24:
How is it
> printing the object as precisely as possible
, if changing the layer_HEIGHT gives me something unpredictable like this?
https://i.imgur.com/ncwMHid.png
https://i.imgur.com/6zhpbx6.png
https://i.imgur.com/1P4vDOp.png

As far as I can understand this, reducing the layer_height, increases the overall width:
https://i.imgur.com/1MwtC0H.png
https://i.imgur.com/VlaI1WV.png

lukasmatena, 2019-04-24:
@hevilp
Thanks for the homework. I don't see any issues. The gcode min/max x-coordinate difference is 24.58mm, which is the distance of the centers of external perimeters. Slic3r tries to make extrusions that are 0.45mm wide, so the actual dimensions of the object will be larger by one half of that on one side and one on the other, giving total width of about 25.03mm. The actual dimensions of the model are 25mm. That's not a bad job I suppose.

@lixxbox 
Assuming the cross-section of the extrusion (half-circle, rectangle, half-circle) places a clamp on what minimal extrusion width can be achieved to `layer_height+nozzle_diameter`. Setting extrusion width lower will result in discarding that extrusion as not printable. That's why it depends on layer height. And it is erratic because you are right on the edge and what will and what will not be discarded depends on small differences on the mesh, where exactly is it sliced (that changes too) and numerical inacurracies. What you are seeing is consistent with the behavior described above.

lixxbox, 2019-04-24:
I don't get it. 
Are you talking about a software limitation or a physical limitation? If physical, would this be more like a theoretical, or a practical one?
I understand, why you try to squeeze the lines, although there are other methods.
I understand, that in a 3D model with fine details or thin walls there is not enough space to properly lay down the lines. 
But I don't understand why it should make a difference to print a part with extrusion_width 0.4 and layer_height 0.1, or with extrusion_width 0.4 and layer_height 0.2? 
I don't understand why you force your slic3r to fail in certain situations, altough there is no reason to.

Have you ever tested what happens when you set the extrusion_width smaller than the nozzle diameter? Maybe with another slicer that sets spacing = extrusion_width (cura, s3d)?
In my adjustment phase I reduce the extrusion_width at a nozzle_diameter of 0.4 step by step down to 0.30, while I closely observe the toplayers. There you can see that the extrusion rate is decreasing and the lines are moving together at the same time.
When the toplayer is closed, I start to adjust the extrusion multiplier.
I don't have any problems with the extrusion_width to go down that far, on the contrary. The extrusion becomes much more even. I tested this with layer_height 0.1-0.25.

Whatever. I understand that you are not willing to change the method, because I can't judge the rat tail behind it. Since I don't want to guess values, but want to set my extrusion correctly (I probably have a different workflow than you), I'll move on. 

Maybe we'll see each other again in the future. :-)

lukasmatena, 2019-04-24:
@lixxbox 
Ok, so what I wrote before was not correct. I was apparently way too fast, sorry. Slic3r leaves the settings to the user and allows to set the extrusion width independently. Changing layer height alone should not stop anything from being generated and if it does, I would look for the problem elsewhere. Could you share the model so we both look at the same thing?

> When the toplayer is closed, I start to adjust the extrusion multiplier.

I'm not that experienced in printing itself, but adjusting the extrusion multiplier is not in my standard workflow. Do you do this with all the slicers? Based on what do you do the adjustments?

> you are not willing to change the method

We will not "make spacing equal to extrusion width" simply because it makes no sense. At least not if "extrusion width" means what it means in Slic3r. The reasoning for changing something this basic and tested would have to be way stronger. I understand there may be other ways of doing it, but the simple change suggested here is not one of them. What would be possible is add an option to calculate the needed extrusion width from the spacing that the user provides, but it might be even more confusing than the current state, which is only confusing to people who want to do the exact fit of certain number of perimeter lines on their model.

lixxbox, 2019-04-25:
> Could you share the model so we both look at the same thing?

It's not my own model. I would have to check the license first, or ask the creator.
Maybe I could create a similiar example later.

> adjusting the extrusion multiplier [...] Do you do this with all the slicers? Based on what do you do the adjustments?

I do this with Cura and S3D.
Mainly I adjust the extrusion_multiplier according to optics. I print small testobjects until reducing the extrusion_multiplier step by step until the toplayer begins to seperate again. Then I take the previous value. I'm also experienced enough to see, where my printer is overextruding. My prints are dimensionally accurate.
The main goal is to achieve uniform extrusion by adjusting the pressure in the nozzle using extrusion_width, extrusion_multiplier, temperature and so on. It's hard for me to describe. English is not my native language.

You're talking about confusion. I get confused when something happens behind the curtain that I can't understand or control right away. First I had to inspect the resulting gcode, make tests etc. Then I found the flow math description, which makes it understandable, but doesn't help me in the end.

I have the following suggestion:
How about you insert a field to specify the spacing _or the overlap factor_. The documentation linked by bubnikv even offers an overlap_factor of 0-1. It's the 1 in the formula: spacing = extrusion_width - layer_height * (**1** - PI/4)

I've prepared something for you.

[image]

lukasmatena, 2019-04-29:
@lixxbox If your workflow consists of first guessing the extrusion width and then extrusion multiplier, you can of course do the same in Slic3r. I am just surprised than someone who does not "want to guess values" works this way:-) The logic of Slic3r was meant to prevent the under/overextrusion.

In future we may add some tooltip or something to do the inverse calculation for you, so that by setting the required line spacing Slic3r suggests required extrusion width to have the extrusions spaced as you want. Something along the lines of the overlap factor you suggest is another option, although one quite difficult to implement and prone to introducing bugs. So we would be reluctant to do that, unless we hear good reasons to do so. It would therefore help to answer my first question: What is the problem you are solving? Do you want to have control over how many perimeters are used without leaving space for gap fill (as it was with #1405)? Is it dimensional accuracy of the prints? Is it the fact that it's poorly documented and confuses user at first? Is it something else? The thread so far did not answer that for me.

lixxbox, 2019-04-29:
"guessing values" referred to
> If you want specific number of perimeters, you should lower the extrusion width

which would throw the complete calibration overboard. No thanks.

The problem with Slic3rs logic is, that it assumes perfect extrusion. This is only theoretical. In reality it depends on several parameters and it would be nice if this could be set on a case-by-case basis.

For me, you don't have to waste any time trying to add a tooltip.
I'm ending this whole Prusa Edition topic for me. No need to explain anything further.

Greetings

beeb, 2019-05-03:
Hey guys, author or #1405 here. I'm glad I'm not the only one who's confused by this "extrusion width" vs "perimeter spacing" thing.

I find the "overlap" concept to be confusing, and it wouldn't be good to have a checkbox that makes `extrusion width = spacing` because of the dimensional accuracy problem @bubnikv mentioned.

To this date, I still find my suggestion of switching from "extrusion width control" to "perimeter spacing control" on demand rather a good one.
With this, the slicer would still behave the same behind the scenes, but the user interface would control the `spacing` variable in this equation instead: `extrusion_width = spacing + layer_height * (1 - PI/4)`. For all intents and purposes, the code might even overlook this calculation and directly use the spacing value since that's what ends up in the gcode finally.

bubnikv, 2019-05-04:
What I personally may find a bit useful is to provide a hint similar to the "Recommended object thin wall thickness". Something like a "Recommended extrusion width for a thin wall thickness at layer height ... " at the "extrusion width" parameter page may do.

bubnikv, 2019-05-04:
Another thing would be to improve the gap fill / perimeter algorithm, so that instead of the last thin gap fill the previous perimeter width and centerline would be adjusted accordingly.

th33xitus, 2019-05-04:
What about @beeb 's suggestion? I really like the idea. And i think introducing bugs shouldn't be much of a problem with this approach?

beeb, 2019-05-04:
> "Recommended extrusion width for a thin wall thickness at layer height ... " at the "extrusion width" parameter page

@bubnikv That would already be a first step for sure, would save us doing trigonometric calculations. Still the "thin wall thickness" needs to be defined. A lot of people are gonna want an even number of perimeters inside walls that are multiples of 1mm, but that would only be true for 0.4 or 0.5mm nozzles. Could the tooltip potentially have a text field so that the user can enter the desired spacing and the text updates with the recommended extrusion width?

bubnikv, 2019-05-04:
> Could the tooltip potentially have a text field so that the user can
enter the desired spacing and the text updates with the recommended
extrusion width?

Likely yes.

I still do not understand your use case. In all but trivial prints, the
wall thickness varies over the print wildly, therefore it does not make
much sense to try to tweak the extrusion width manually to fit any of these
constraints. This should be handled by the gap fill. Also the object is
shaped mostly with an external perimeter, and it must be set neither too
low nor too high, otherwise the print accuracy will suffer.

th33xitus, 2019-05-04:
The extrusion width affects the top layer of a print as well as the perimeters, being able to tweak it makes you able to print absolutely watertight and have gorgeous looking top surfaces which are perfectly printed with the right amount of material. No under extrusion and no over extrusion. Im not speaking of looking at a print from 1 meter distance, im speaking of looking at a print under a microscope.

beeb, 2019-05-05:
> In all but trivial prints, the wall thickness varies over the print wildly

Well trivial prints but also a lot of functional parts. Often times, these parts have mostly simple geometries and designers use round millimeter values for a lot of features. It can dramatically increase the print time if gap fill is extensively required. Features that have a constant section (like often found in functional parts) will require this additional gap fill for each layer! I'm happy to provide example parts but I think it's easy to grasp this concept.

mroek, 2019-10-09:
I completely agree with @beeb on this one. I find extrusion width in PrusaSlicer to be pretty useless. I want to be able to control spacing instead, as in most (all?) other slicers. There is nothing preventing PrusaSlicer from doing the necessary math (like other slicers) to calculate the proper extrusion amount if the input parameter is given as spacing instead. And that would also make it independent of layer height, eliminating the need to tune that parameter every time you change layer height (to get the spacing you want).

Why have an input parameter that doesn't in any way intuitively and directly affect the output? I design most of the stuff I print myself, and I usually design stuff so that walls have a thickness that is a multiple of 0.4, and a height that is a multiple of the intended layer height. This works as expected with S3D, which is what I currently use, but I am getting tired of the exceptionally slow update rate of that slicer, so I've been looking into PrusaSlicer, which has a lot going for it.

My main reason for using multiples of 0.4 mm for wall thicknesses is of course to avoid unnecessary gap filling, and thus save on print time.

aminerhayem123, 2023-04-06:
do you guys have any code to help me calculating the lowest thin wall  thickness in a 3D model i'm using python

closed 2026-09-29 by prusaslicer-bot[bot]