prusa3d/PrusaSlicer #2164: Brigher colours for previews

guestisp, 2019-04-24:
I haven't seen any possibility to create custom color scheme, so I'm asking here: would be possible to use a brigheter color scheme for the preview mode ? S3D and Cura has both a much ligheter (and easier to see) color scheme.

In the same way most application has a "light" or "darker" color scheme, based on user preferences, a brighter (or darker) preview schema would be usefull, mostly on monitors with bad contrast ratio.

guestisp, 2019-04-29:
I was trying to change a couple of colors, just as test, but the written color seems to be set as darker in the preview.

Is anything that make the whole preview darker ? If yes, how can I change it ? I would like to see if the overall visiblity will improve in my monitor

guestisp, 2019-04-29:
Even the gap fill color, that should be totally white ('FFFFFF'), is not white by greyish

bubnikv, 2019-05-04:
Thank you for your suggestion. Please provide a pull request.

guestisp, 2019-05-04:
I was trying to do this, just a question: there is "something" that makes all color darker. If I set "FFFFFF" as color, it will be shown as light gray, but not white. Can you point me to the code that currently make the color darker?

Panayiotis-git, 2019-05-04:
I think that the models are lighted by a virtual spot of light. So, they are not equally colored since shadows are applied.
Lighter colors should be applied with attention, otherwise the sides of the models will not be distinguished.

guestisp, 2019-05-04:
Sure but current color schema is too dark, on certain monitors is not very well contrasted.

Seems like a low-bright image

ManuGithubSteam, 2019-07-31:
Yes i totaly agree its very hard to distinguish the speeds in the model because of the blue tones who fade into each other.

jonczeck, 2020-09-14:
[image]

I've made it brighter so you can distinguish all the blue tones. The darkening is happening in the shader. This is a very noticeable visual change. Do Prusa people actually want this? I can make a PR if so.

graizzaro, 2020-10-29:
I found that travel color is too dark to see it on black bed color.

To understand where travels are I had to change point of view and look them against light sky.
Only in this way I can understand travels and change objects position to improve them.

I think that having possibility to change bed or travel color will be a great improvement of the program.

foreachthing, 2020-10-29:
> [image]
> 
> Do Prusa people actually want this? I can make a PR if so.

Make it optinal! 
Long time ago, I made a Solarized theme (optional) for Slic3r... I never found time nor the correct files in PS to make a new theme for it. I don't like the default colors.

n8bot, 2021-11-29:
This might interest you: 
https://github.com/n8bot/PrusaSlicer/releases/tag/version_2.4.0-beta2-release1-n8

Forgive me for following screenshot the spam.
[image]

[image]

[image]

[image]

[image]

[image]

[image]

bubnikv, 2021-11-29:
Great, I see that after SuperSlicer we will get a N8BOT slicer.

P-C-R, 2021-11-30:
@bubnikv  Would this be worth  a PR? I am using the version of @n8bot since a few days and its really an improvement.

bubnikv, 2021-11-30:
Don't know. We need to evaluate. Are these supernew n8bot colors hard coded? Frankly I think most of us are happy with the current color scheme. Do we want the colors chemes to be configurable? And why? Is there a risk that everybody will adjust the colors differently and then if two screenshots are taken of G-code viewer, the colors will be so messed up, that it will become a burden in cooperation because on each screenshot a color will have a different meaning?

P-C-R, 2021-11-30:
my 2C are that it would be nice to have a options for changeing the colors to another theme. The argument with the "different" meaning is right. But in the end you have the legend. And for cooperation its always best to load the gcode into the the own slicer.
A lot of CAD programs have the options to to change the color of sketches and lines etc.

n8bot, 2021-11-30:
I will try today to bring the colours closer to the stock colours and see how it looks.

The problem with the stock colours is that they seem to be haphazardly chosen and do not have even luminance values. Rather, they don't *need* to be even, but they need to be... considered, when chosen.

I think you will find that many users are not happy with the current prusaslicer visualization, including the colours. The colour choices are not offensive or anything, just not optimal.

I will experiment with adding a preferences switch to change the colours, too... but I am also reluctant to have multpiple colour choices for the reason bubnikv mentions: the colours should have a consistent meaning across screenshots, versions, etc.

mikolaszuza, 2021-12-02:
@n8bot I applaud the effort!

>  but they need to be... considered, when chosen.

I am the one who picked the current colors. My priority was clarity while trying to stay close to the previous color scheme. People hate change and are used to what the colors mean, so I didn't want to make life harder for them.

I like what you are trying to do. Please don't take my feedback negatively, just sharing my honest first impressions.

The colors seem very solarized
[image]

PrusaSlicer left, n8 right
I find our current scheme much easier to read when inspecting layer-by-layer. The blues and violets blend together and are hard to distinct for me.
[image]

I am so used to RGB <-> XYZ when coloring axis, not a huge fan of changing that. Was this to accommodate people with some type of color blindness?
[image]

n8bot, 2021-12-02:
Thank you for taking the time to consider this and provide such thoughtful feedback.

> I am the one who picked the current colors. My priority was clarity while trying to stay close to the previous color scheme. People hate change and are used to what the colors mean, so I didn't want to make life harder for them.

**_I hope I did not offend you with my description of the current colours._**

> 
> I like what you are trying to do. Please don't take my feedback negatively, just sharing my honest first impressions.
> 
> The colors seem very solarized [image]
> 

**_This is the lighting, not the colours that is "solarizing" things. It's not polarization, it's just specular highlights from the new lighting setup I made. I have separated the colours and the lighting changes -- the lighting changes, are, IMO, much more important than changing the colours. In fact, I'm not submitting the colours as a PR! I summitted the lighting improvements as a PR, because they are a more sensible first step in correcting the current colour problems in PrusaSlicer (and, no offense, there *are* problems.)_**

> PrusaSlicer left, n8 right I find our current scheme much easier to read when inspecting layer-by-layer. The blues and violets blend together and are hard to distinct for me. [image]
> 

**_I agree with you here. Some of this is due to the lighting (I improved it last night, for exactly this scenario of looking downward at an angle... I'm not sure if you are showing photos with the newest lighting setup (I did not release the binaries for the newest changes, only committed them to my branches._**

**_Some of the colours can definitely be swapped around if need be... or not chosen at all! (Remember, I didn't submit this as a PR? :P)_**

> I am so used to RGB <-> XYZ when coloring axis, not a huge fan of changing that. Was this to accommodate people with some type of color blindness? [image]

**_This is precisely the reason. This *IS* RGB, but for colorblind people. Notice how the colours are red-ish, green-ish, and lue-ish?_**

This is an accessibility-first colour scheme. IMO, accessibility for colours needs to come first (and only!). If "colorblind safe" colours are relegated to a special setting hidden in menus, the colorblind uses are left in the dark when it comes to *other* users posting screenshots of their slicer setups, etc. In order for accessibility like this to work, it has to be adopted by all. In a sense, we need to accommodate the colourblind by offering compromise for our own colour preferences.

But, yes, these colour changes are radical. If radical change is to be made, the time is now. If any change is to be made, the decision must be made: should it be radical, or subtle? IMO, it's time for an attention-grabbing radical change. This will help with marketing, and also show that the Prusa team cares about accessibility for all users.

I am not colourblind or anything, I just feel that the "accessibility first" mindset is a productive one.

So, last but not least, what I see as the "problems" with the current stock PrusaSlicer colours:

1. First, duh, they're not *exactly* colourblind safe (but honestly, they're not horrible.)

1. The range gradient is one of the least useful (albeit most popular) spectrums in use. Rainbow spectrum is kinda bad. The luminance values are all over the place, and the colour changes are not logical unless seen together (the sharp transition from dark blue to bright yellow makes no sense without context).

1. The luminance values, in general, are all over the place. Some colours are very dark, some are ridiculously bright and over-saturated. (This is without talking about the lighting situation -- this is the RBG values of the colours themselves I'm speaking of.)

1. The last thing, which is slightly unrelated to the colours, is the *lighting* of the colours. Which is much more important, and targeted by my PR https://github.com/prusa3d/PrusaSlicer/pull/7392 Notice in the stock PrusaSlicer visualization, the colours appear much darker in the scene than they do in the legend?

n8bot, 2021-12-02:
As a side note, based on your feedback, I switched up my new colour scheme slightly. I made the internal infill pink again (my original choice) to be closer to the prusaslicer stock red. (I thought people would be like YUCK PINK!!?) I like pink,

I also swapped the support material colours around because the green does not contrast enough with the orange when viewed as a colourblind person (according to photoshop). Photoshop won't let me screencapture their simulated colour blindness so you'll have to try it yourself or ask a colourblind person lol.

[image]
[image]

So, the pink colour brings the scheme *closer* to stock colours, while the support colour changes bring them *farther* lol. This is difficult for sure, making a compromise.

I haven't released binaries of these changes yet, FYI, but they are committed to my repo's main branch (`n8-240b2-release`).

foreachthing, 2021-12-02:
I followed this "thread" with awe and, this is my personal opinion, PS should be customizable!
I'd like to have an xml (or whatever means) to fully customize the colors. Then the themes are listed in the options and can be chosen from.
_This_, what @n8bot has done, could be a predefined theme - I love it and I'm off to fetch his branch.

n8bot, 2021-12-02:
> I followed this "thread" with awe and, this is my personal opinion, PS should be customizable! I'd like to have an xml (or whatever means) to fully customize the colors. Then the themes are listed [...]

Thank you for the kind words!

As an added example of the colourblind simulations, with the range scale (like for speed, width, flow, etc).

In the rainbow spectrum, it is not exactly clear the different meanings of the different colours, without referencing the rainbow in your mind. or looking at the legend. The idea behind the magma spectrum is that it provides a natural gradient that is more obvious to interpret, and especially for colorblind people, to accurately see the most of the spectrum.

**Stock:**
[image]
[image]
[image]

**Magma spectrum:**
[image]
[image]
[image]

n8bot, 2021-12-02:
For reference, here is the chart showing the stock prusaslicer colours on top and my colour scheme on the bottom:
[image]

The markings are just for me to keep track of which colour is which and reused where. (Hours of staring at colours makes eyes hurt and colours hard to distinguish lololo.)

n8bot, 2021-12-02:
Sorry for all the mixing around. "Option 2" is now settled and committed to my main branch at `n8-240b2-release` in case anyone wants the tiny changes lolo... I tinker now, so I never have to tinker again.

foreachthing, 2021-12-03:
@n8bot I'm sure you know this site: https://ethanschoonover.com/solarized/.
I always liked those colors (_I added that solarized theme to the original [Slic3r](https://github.com/slic3r/Slic3r)_)! They're also easy on the eyes. Got TeXstudio and notepad++ setup like that and I wish more programs could adopt that solarized

n8bot, 2021-12-03:
I'm not sure if I've seen that particular site, but thank you! That is a lovely resource.

n8bot, 2021-12-03:
I love the ideas behind the "solarized" theme (not to be confused with the photographic term solarization which was brought up earlier in this thread, lol).

The background colour theories are amazing. I love the monotone ranges for content and background. That is truly brilliant.

I think the accent colours could be adapted to be better suited for colour blind people. Only small tweaks would be needed, I think. Currently, the red highlight colour loses its meaning entirely when seen by a colourblind person.

[image][image]

[image][image]

Now, I'm sure the colour theme I have put into the slicer visualization could be criticized in similar ways, though I've tried to mitigate it. There is no perfect solution, at all.

What I would like to see, perhaps, is a blend of the "solarized" theme with colourblind safe theory, like the stuff briefly mentioned in that [Adobe blog page](https://spectrum.adobe.com/page/color-for-data-visualization/). There is already a lot of overlap and similarity between the two schemes.

dartrax, 2021-12-04:
> I am the one who picked the current colors. My priority was clarity while trying to stay close to the previous color scheme. People hate change and are used to what the colors mean, so I didn't want to make life harder for them.

That's exactly the point why I think it makes sense to have some predefined themes to choose from. I assume many people coming from other slicers (like me, switching from S3D) will feel more home when they see colors that they are used to in the preview. I would love to see a preset with the colors of S3D, others might like to have (or contribute) a Cura preset. 

> I followed this "thread" with awe and, this is my personal opinion, PS should be customizable! I'd like to have an xml (or whatever means) to fully customize the colors. Then the themes are listed [...]

+1

theturtle32, 2022-05-31:
I stumbled upon this while Googling the phrase "prusaslicer feature color scheme."

I'm *loving* PrusaSlicer, coming from Simplify3d, but have been finding myself wishing for S3D's clearer, higher-contrast feature-based color-scheme. I initially thought the colors might be configurable, like in an IDE, and upon not finding anywhere in preferences to change the color scheme, went searching.

Definitely +1 for the thoughtful changes being discussed here!

whi-tw, 2022-08-26:
I've been searching around for something to make the 3D scene more visible when using, say, black filament color.

I was tinkering with the idea of implementing Gooch Shading [^1] [^2], but just can't get my head around this codebase.

[^1]: https://en.wikipedia.org/wiki/Gooch_shading
[^2]: https://www.youtube.com/watch?v=tvKLXbhVBnw

BrockPlaysFortnite, 2023-04-21:
Same here superslicer lets you change a .ini file and add your own color codes to it

TheNeg0t1ator, 2024-04-21:
> I've been searching around for something to make the 3D scene more visible when using, say, black filament color.
> 
> I was tinkering with the idea of implementing Gooch Shading [1](#user-content-fn-1-ebe02adff78d8777e5848e09f045ebf1) [2](#user-content-fn-2-ebe02adff78d8777e5848e09f045ebf1), but just can't get my head around this codebase.
> 
> ## Footnotes
> 1. https://en.wikipedia.org/wiki/Gooch_shading [↩](#user-content-fnref-1-ebe02adff78d8777e5848e09f045ebf1)
> 2. https://www.youtube.com/watch?v=tvKLXbhVBnw [↩](#user-content-fnref-2-ebe02adff78d8777e5848e09f045ebf1)

I think this is done by checking the luminance of the part of the object? 
the only problem i see, is that the shading must be done on the whole object, rather than the seperate lines, as this is a good reason for them to be less bright.
i think this would be a funky way to showcase a file in the slicer.
should check out [this pr](https://github.com/prusa3d/PrusaSlicer/pull/7392) as well, it changed the colours. 
Is there an issue open for the gooch shading?

closed 2026-09-29 by prusaslicer-bot[bot]