prusa3d/PrusaSlicer #2165: Infill before perimeters starting at layer

guestisp, 2019-04-24:
I've seen that printing the infill before perimeter allows me to print faster and produce smoother walls. This because (i think) any roughness caused by an higher speed during the infill, is "flattened" when printing slowly the perimeters. In the oppsite way, you start with a smooth surface but then any trouble on the infill will make the surface worse.

Cura prints in this way as default, but only after a certain amount of layers (2, I don't remember if this is configurable)
Something similiar would be useful in slic3r, as the very first layers should be printed starting from the perimeters, for a better adesion on the bed.

supermerill, 2019-04-25:
Problem with infill is that they sometimes rely on the inner perimeter to attach themselves to the part.
Also, the idea is that the inner perimeter is fixed, the outer ones are "pushing" against it to the external side and the infill lines are "pushing" against it to the inner side.

Imo, i think you should bring more than "I've seen that printing the infill before perimeter allows me to print faster and produce smoother walls" as proof. For exemple, you can link a model, the profile you used, the "manually" modified g-code with infill before perimeters, and some pictures to show the improvement. With that, we can try to replicate and we know what to look at.

guestisp, 2019-04-25:
If you start to print the infill in reverse order (first infill then perimeters) after the first layer, the infill can stick to the previous printed layer, there is no need to make it stick to a perimeter. That's why cura reverse the infill printing starting from layer 2 or 3

Additionally, even the overhang perimeters would be' better as they can stick to the already printed infill plus a little bit of the previous layer.
That's why printing perimeters from inside to outside is better for overhangs. In this case you also print the infill, so even more inside to outside...

supermerill, 2019-04-25:
> there is no need to make it stick to a perimeter.

So you print with 0% overlap? If not, you should try it, as it may resolve your issue.

guestisp, 2019-04-25:
I'm using 25% as overlap.
What i meant is that the infill doesnt need to *only* stick to a perimeter to adhere. It can *also* Stick to the previous layer

Only in the first layer the infill must adhere to a perimeter as you don't have any previous layer to stick to

guestisp, 2019-04-26:
> Problem with infill is that they (sometime) rely on the inner perimeter to attach themselves to the part.

Reversing the print order starting at a certain layer, in example layer 2 or 3 or whatever (but not the first layer obviously) like Cura does, allow to infill to stick to the previous printed layer, so you don't need to rely on the inner perimeter to attach to the part.

> Imo, i think you should bring more than "I've seen that printing the infill before perimeter allows me to print faster and produce smoother walls" as proof. For exemple, you can link a model, the [...]

I don't have a print right now, as I'm doing tons of testing with multiple slicers. I've found that Cura, in example, is able to print better with a smooth surface finish, then i've set Slic3r to print the infill before the perimeters and the same smoothnetss I have with cura was also archieved with Slic3r. Maybe also related to the seam position ? Printing in reverse order place the seam internally ? I don't know.

I can post a pic where two benchies were printed with the same settings except the print order. (the filament is changed, but that's not the issue)

Cura reverse the print order at layer 3, the first 2 layers are printed starting from the perimeters.
Slic3r prints everything in reverse order (if "infill before perimeters" is set) thus there are some issues in getting the first layer sticking to the bed 

What I'm asking is something to set the layer number where reversing the order (or a new flag like "Infill befor perimeter starting at layer 2") so that the first layer is always printed in standard way

supermerill, 2019-04-26:
> What I'm asking is something to set the layer number where reversing the order (or a new flag like "Infill befor perimeter starting at layer 2") so that the first layer is always printed in standard way.

Sure, It's easier than making a modifier cube for the first layers each time.

guestisp, 2019-04-26:
With the modifier cube you can't change the print order

supermerill, 2019-04-26:
ah, yes, it's only on my fork, sorry.

guestisp, 2019-04-26:
Do you have a compiled Linux version? I would like to test your fork

supermerill, 2019-04-26:
> Do you have a compiled Linux version? I would like to test your fork
nop (not yet).

> I think you made some improvement related to the seam right? I've read wrong?
I just added a new option to avoid travels. But nothing yet to reduce the seam.

guestisp, 2019-04-26:
> I just added a new option to avoid travels. But nothing yet to reduce the seam.

IIRC, you also have an "hidden" mode for the seam. Do you know why most of your fixes aren't backported to the Prusa version ? They are really usefull, in example, the hidden seam, the extrusion width % based on nozzle size, continuous loop (one of my biggest issue) , ironing.............

The code is already done (by you) , why not merging it in the official repo ? I don't understand this......

vojtechkral, 2019-04-26:
> The code is already done (by you) , why not merging it in the official repo ? I don't understand this......

Merging the code into the official repo means taking responsibility for the code, which implies, for example testing, verifying it works on different OSes/architectures etc. It also means maintaining the code in the future. At the very minimum, someone has to at least _read_ the code.

guestisp, 2019-04-26:
> At the very minimum, someone has to at least read the code.

That's for sure. But there are some useful feature with some issue pending from many month/years that are already done and working on a fork. I think would be faster to merge something already done than developing the same thing totally from scratch.

supermerill, 2019-04-26:
>  I think would be faster to merge something already done than developing the same thing totally from scratch.

Faster in the short term, probably. But someone may want to write a better, less error-prone and more maintenable code.

> IIRC, you also have an "hidden" mode for the seam.

Not really, i just renamed "nearest" to "hidden". Bad decision on my part, I will rename it again to "nearest hidden" for the next release.

GeorgZiegler, 2019-05-05:
As a Feature Request, what i would recommend as an improvement :
(Only) for 1st layer print in following order : Skirt > Bottom Solid Infill > Perimeters > External Perimeters.
So basically 'Infill before Perimeters'. In my opinion this would be an improvement, because the
solid infill sticks better to building plate than a single perimeter line. 
A single perimeter line is often 'wrapped up'. By reversing the order for 1st layer the perimeter lines have
a better chance to stick to the already printed infill lines besides them.

guestisp, 2019-05-05:
Agree

rongith, 2019-05-20:
i wouldn't like "Bottom Solid Infill > Perimeters" for my use. that would be not nice for some filaments like colorfabb xt which are thick and stick well to itself. i already decreased acceleration on first layer to 300 to be sure the infill doesn't curl up when the nozzle do a 180°.

rtyr, 2021-01-28:
I don't see a need to implement this. I have also the opposite experience with the smoothness of the external walls, when the infill is printed first. Same thing is actually mentioned in the Cura, where this setting is no longer enabled by default.

rtyr, 2021-08-12:
No response. Closing.

closed 2021-08-12 by rtyr