prusa3d/PrusaSlicer #2166: Filament selection inconsistency between Slic3r and printer

tobymurray, 2019-04-25:
### Version
1.42.0-beta2+win64-full-201904140830

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3s firmware 3.7.0-2201

### Behavior
Not sure where the problem lies, or if it's an integration issue between Slic3r PE and the firmware. When I select "Original Prusa i3 MK3S MMU2S Single in Slic3r PE, then go to `Print Settings`, under Extruders I have the option to change the extruder number. In this case I changed it to 4 for everything, because I wanted to use the filament loaded in filament slot 4. When I go to print the file, I'm given the opportunity to select the filament at that time. It seems like regardless of what I select, the filament in position 1 prints (I tried "filament 4" and I tried "filament 1", both yielded filament 1).

Steps:
1. In Slic3r PE, select MK3S MMU2S Single for the printer
1. Navigate to `Print Settings` in the top bar
1. Navigate to `Multiple Extruders` in the side navigation
1. Change the all the value in `Extruders` to something besides 1 (e.g. 4)
1. Slice and export the g-code
1. Print the file
1. Note you're prompted for the filament to use, select anything besides 1 (e.g. 4)

* _Expected Results_
Filament 4 will be used, either respecting the Slic3r selection or the printer selection

* _Actual Results_
Filament 1 is loaded

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
So far as I can tell, literally any file sliced exhibits this behavior.

This may be "fixed by documentation" or a tooltip or something, as a new user of an MMU it's not at all clear how selecting an extruder in Slic3r PE relates to selecting a filament on the printer. The current slic3r docs have:
> For single material prints, where the secondary extruder is to be tasked with a particular extrusion, the Multiple Extruders section of the Print Settings tab gives the ability to assign an extruder to each extrusion type.
which doesn't make much sense to me. Why would one use multiple extruders for a single material print? Different nozzle sizes?

bubnikv, 2019-04-25:
If you peek into the printer settings -> start G-code, you will find a Tx code there, which instructs the printer to ask for a printer.

Also in the printer settings the printer has only one extruder assigned, therefore all extruder settings are essentially ignored. The printer & print profiles behave as if you printed on a single extruder / single filament machine with the exception of the start G-code, which triggers the printer to ask the user to select the filament and then the start G-code instruct the printer to load the filament from the MMU2 into the extruder before it starts printing.

bubnikv, 2019-04-25:
The print and printer profiles are independent, one print profile may be used with multiple printer profiles. That is why the print profile editor allows you to assign extruder IDs, which are not existent on the active printer profile. We may think of changing that, I agree that this is confusing, though it asks for a little thinking of how to do it right.

tobymurray, 2019-04-25:
So it sounds like just user error/area for improvement in documentation, right?

> therefore all extruder settings are essentially ignored

Does "essentially" mean in some capacity they are not ignored? As a new MMU owner, I'd definitely recommend hiding these in single-extruder mode if possible.

> The printer & print profiles behave as if you printed on a single extruder / single filament machine with the exception of the start G-code, which triggers the printer to ask the user to select the [...]

Awesome, that's roughly what I was hoping. Selecting the extruder to choose the filament didn't seem right. Is this at odds with actual multi material printing though? In that context when I want to choose a separate filament I _do_ assign a different extruder, right?

> The print and printer profiles are independent, one print profile may be used with multiple printer profiles. That is why the print profile editor allows you to assign extruder IDs, which are not existent on the active printer profile.

Sorry, you've completely lost me here... can you dumb it down a bit? Is "the print" the generated g-code? What's a print profile vs. a printer profile? Is the printer profile the selection of the printer and the print profile everything else?

bubnikv, 2019-05-04:
print vs. printer profiles refer to the content of the print vs. printer parameter tabs. They reflect the content of the files stored in Slic3rPE/print vs. Slic3rPE/printer folders.
(you will not find the system profiles stored there, they are loaded all from a single Slic3rPE/vendor/PrusaResearch.ini file)

xoxota99, 2019-05-24:
Reading through all of this, what I'm understanding here is "Slic3r in MMU2S Single mode will always only use filament #1", regardless of what's in the "Print Settings > Multiple Extruders" tab (ignored during Single mode?), or the Tx code in the "Printer Settings > Start G-code" box (printer will ask you for an extruder, then ignore whatever you select?).

Is that correct?

bubnikv, 2019-05-24:
>Reading through all of this, what I'm understanding here is "Slic3r in MMU2S Single mode will always only use filament #1", regardless of what's in the "Print Settings > Multiple Extruders" tab (ignored during Single mode?)

that is correct.

> or the Tx code in the "Printer Settings > Start G-code" box (printer will ask you for an extruder, then ignore whatever you select?)

Slicer knows nothing about what is happening on the printer. So Slicer will use parameters of the one and only filament profile active for the MMU - Single printer profile, and it will generate a G-code for the printer, where this Tx is included. The rest is handled by the printer, it will ask the user and load one of the filaments through the MMU unit.

PxT, 2019-06-21:
So why is this Tx code generated in Single profile mode if, as you say, the printer will always use Filament 1.  What purpose does the "choose filament" menu serve? If it does nothing it should not be displayed. This is very confusing for the end user.

bubnikv, 2019-06-24:
> So why is this Tx code generated in Single profile mode if, as you say, the printer will always use Filament 1? What purpose does the "choose filament" menu serve? If it does nothing it should not be displayed. This is very confusing for the end user.

In the "single color for MMU2" mode.  Slicer will not generate any other T command other than Tx at the start of the print. It is up to the printer to select the filament.

PxT, 2019-06-24:
This is incorrect.  I choose "Filament 2" on the menu, printer still uses filament 1 for everything except the purge line.  Checking the GCODE I can see that there is a T0 command in there.  Gcode was generated with printer_settings_id = Original Prusa i3 MK3S MMU2S Single

tobymurray, 2019-07-11:
While it's not consistent (I have been able to select the filament) I see this as well. Regardless of my selection on the printer, the printer selects filament 1. I don't know what context leads to this, but it was both confusing and frustrating (and what ultimately lead to me opening this issue in the first place).

jbcantrell, 2019-07-25:
I can verify this behavior.  PrusaSlicer is inserting a T0 command well after the Tx command.  No matter what Filament I choose on the printer, it, after doing the initial bed leveling, it will load filament 1.

KennyTK, 2019-07-31:
I've run into the same issue while using PrusaSlicer 2.0.0.
Regardless of which filament I choose on the printer, it will always cycle to filament one.

bubnikv, 2019-08-01:
Please provide a 3MF and repeat your issue with each of the 3MF provided. I am afraid there are apples mixed with oranges in this issue and we want to avoid more confusion. There may be outright bugs in slicer, there may be outright bugs in the firmware, there may be a misunderstanding at your side as well.

Impruve, 2019-08-12:
Ok.. so.. I think i know exaclty how this happens, Its PrusaSlicer Fault.

This always happens IF you have the software open with the profile MMU2S active, then change to MMU2S Single and then slice the file.

What you need to do is change to Single, close PrusaSlicer and re-open, the sliced file will no longer change to Filament 1 when you start a print with any other filament.

Now... im not sure if this only happens after i sliced a file for MMU2S and then i change to Single to slice a one color print but one thing is certain, if i close the software between profile changes i never get the problem.

lukasmatena, 2019-08-21:
@Impruve Thanks, this was helpful and explains why we had such issues reproducing it. You are right. Switching printer profiles from MMU to MMU Single causes it. The printer would ask which filament to choose only to override it with filament number one soon after. This should now be fixed by https://github.com/prusa3d/PrusaSlicer/commit/40d313961e474fce91e0e94858254f149e877d91. The fix will be part of following release 2.1.0.

The problem of ignoring extruders from Print Settings->Multiple Extruders is unrelated and in fact not a bug. This was already explained by @bubnikv.

I'm closing the issue. Please, test the behaviour in the upcoming release and let us know if it ok.

closed 2019-08-21 by lukasmatena