prusa3d/PrusaSlicer #2167: Slic3r crashes if the user deletes all the models while in layer editing

lf-, 2019-04-25:
Deja vu? This might be a regression of #193 but a new bug since it looks like the last one was in perl.

### Version
_Version of Slic3r Prusa Edition used goes here_
1.42.0-beta2+win64

Note: it's annoying that you can select everything in the About dialog *except* the version.

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 1903

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3

### Behavior
* _Describe the problem_
An "unhandled unknown exception" occurs if you press the delete all button in the toolbar while in layer editing

* _Steps needed to reproduce the problem_
As above, import a model, click on it, press layer editing, then press the delete all button

* _Expected Results_
Deletes all the models and exits layer editing

* _Actual Results_
[image]

_Is this a new feature request?_
no

FidelCapo, 2019-04-25:
Thank you for report. This issue should be fixed in next release.
Closing

FidelCapo, 2019-05-15:
This issue should be fix in PrusaSlicer-2.0.0-rc.
Closing

closed 2019-05-15 by FidelCapo