prusa3d/PrusaSlicer #2168: unable to compile on linux with SLIC3R_WX_STABLE=1

cabbey, 2019-04-25:
### Version
git master ( >= 82573390c7c1b918ad4d2d7f9837984aca20316b, at least <= 065448e9e5f63f412ac59c8fbbea560f2cccf77e current as of writing this. )

### Operating system type + version
Linux - Ubuntu 18.10

Note that to configure on this system `cmake -DSLIC3R_WX_STABLE=1` is needed. 

### 3D printer brand / version + firmware version (if known)
Not Applicable

### Behavior
Unable to build since 82573390c7c1b918ad4d2d7f9837984aca20316b landed. The new code added in that commit references a function that does not exist, I'm guessing it's something "new" in the "unstable" version of wx:

```
/home/cabbey/src/Slic3r/src/slic3r/GUI/wxExtensions.cpp: In member function ‘void ModeButton::focus_button(bool)’:
/home/cabbey/src/Slic3r/src/slic3r/GUI/wxExtensions.cpp:2332:57: error: ‘class wxFont’ has no member named ‘GetBaseFont’
     const wxFont& new_font = focus ? font.Bold() : font.GetBaseFont();
                                                         ^~~~~~~~~~~
```

#### Project File (.3MF) where problem occurs
Not Applicable

dafoxia, 2019-04-27:
Same problem here.

GetBaseFont() is'nt in wxWidgets 3.0
GetBaseFont() need at least 3.1.0
:(

YuSanka, 2019-04-29:
Thanks for your report. It will be fixed for the next release.

cabbey, 2019-04-29:
confirmed, thanks @YuSanka

closed 2019-04-29 by cabbey