prusa3d/PrusaSlicer #2169: Custom bed texture

guestisp, 2019-04-25:
I'm trying to find a way to create a custom bed texture, like the Prusa original one. Any help about this ? Is this possible or currently only Original Prusa printers are allowed to use a custom bed texture ?

In Cura is possible to create a custom platform for all printers, not only the ones from Ultimaker

supermerill, 2019-04-25:
Last time i checked, you have to change a substantial part of the source code for that.
I tried to create a patch to allow it to grab the file from the profile, but the codebase has changed again the next week. I may will look at it again in the future, when things will be a bit more stable.

If prusa is willing to implement this feature, i will be grateful for that.

guestisp, 2019-06-17:
Fixed in this PR https://github.com/prusa3d/PrusaSlicer/pull/2496

I hope will be merged anytime soon

bubnikv, 2019-08-01:
Implemented in the upcoming PrusaSlicer 2.1.0-alpha1.

closed 2019-08-01 by bubnikv