prusa3d/PrusaSlicer #2170: setting speeds to auto sets them to about 1E-310

SebJez, 2019-04-26:
### Version
1.41.3+win64

### Operating system type + version
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 10.0.17763.437] 

### Behavior
After setting extrusion speeds to auto in the _Print Settings_ tab (which is also represented by 0.0000 in the interface), the actual printing speeds saved to the config and appearing in generated Gcode are small, positive floating-point numbers on the order of 
* _Steps needed to reproduce the problem_
  Enter `auto` as printing speed:
[image]

* _Expected Results_
Slic3r sets the speed to something reasonable like 40mm/s

* _Actual Results_
Sli3er sets the speed to about 1E-310
   * in the Gcode:
      ```M140 S0 ; set bed temperature
      ; PURGING FINISHED
      G1 X103.461 Y100.618
      G1 F1.19073e-310
      G1 X103.461 Y101.118 E3.53530
      ```
    * in the tab
      [image]
    * in _config.ini_
      `perimeter_speed = 1.98454e-312`
      `support_material_speed = 1.98454e-312` etc.

[files.zip](https://github.com/prusa3d/Slic3r/files/3119380/files.zip)

_Is this a new feature request?_ *No*

SebJez, 2019-04-26:
As the first layer speed is corrected to 30 mm/s this causes the printer to effectivelly stop at the beginning of the second layer

x0rtrunks, 2019-04-28:
I don't think Auto speed has worked in quite some time. At least the volumetric limit still seems to apply.

manticorp, 2019-05-01:
Don't you need to enter ```0``` for auto speed? Not ```auto```? When I hover over the field in my edition of Slic3r, that's what it says?

rtyr, 2021-01-20:
It is not possible to use "auto" value in the current 2.3.0.  Use `0` for auto speed. Closing.

closed 2021-01-20 by rtyr