prusa3d/PrusaSlicer #2171: Sequential Printing not working

iyoman, 2019-04-26:
### Version
1.40.1

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
ADIMLab gantry new version

### Behavibe
I have a model I'm trying to print sequentially, but it gives me an error: "Config. not valid: some objects are to tall and cannot be printed without extruder collisions" This even though the parts would clearly not hit each other. I've messed with the size of the extruder and everything.

reloxx13, 2019-04-28:
post a screenshot of ur object positions and ur extruder offsets.

iyoman, 2019-04-28:
Will do.

agryson, 2019-07-12:
I am having this issue with an mk3s (stock) with the stock profiles in PrusaSlicer - only change to default profiles was checking "Complete Individual Objects" :
[image]
[image]

Despite > 50mm distance between parts for an extruder clearance of 45, the part is 47mm high

agryson, 2019-07-12:
Note to above,  this behaviour persists regardless of if part 1 is at the front of the plate or the back (I thought maybe it was a collision detection of the X axis but apparently not)

agryson, 2019-07-13:
This seems to be due to the max height issue in #1095

agryson, 2019-07-13:
Tested : The issue no longer appears if I ensure all objects are below the printer's Clearance height

Mazhive, 2019-11-01:
Why cant there be an option that features a pause long enough to take it off and continue with the print so you don't have this issue even with the higher models not having clearance. This would be a great feature i think.

bubnikv, 2020-02-04:
The original issue has been resolved. Closing.

closed 2020-02-04 by bubnikv