prusa3d/PrusaSlicer #2173: [Feature Request] Adaptive slicing

guestisp, 2019-04-26:
Variable layer height is awesome, but the original Slic3r has an adaptive slicing feature that automate the process. https://manual.slic3r.org/expert-mode/variable-layer-height#adaptive-slicing

Would be possible to backport this great feature in Slic3r Prusa Edition ?

bubnikv, 2019-04-26:
Duplicate of https://github.com/prusa3d/Slic3r/issues/466

Please, please, very much please, please so much very much extremely much, do your homework and search for the tracker for existing issues before issuing a new one. Thank you very much. Please, please, please, don't create any new issues without first going through the issue tracker. I want to spend my time with my family instead of closing duplicate issues on the issue tracker. Please, do your homework. Please, thank you very much.

closed 2019-04-26 by bubnikv

guestisp, 2019-04-26:
It's not my fault if you don't properly tag the issues. It's a 2 years old issue, with no tag set and with no official updates. If you don't want duplicate issues, please properly handle the existing one or publish a roadmap or something similiar.

Maybe @supermerill has some plan to backport this feature in it's fork

> I want to spend my time with my family instead of closing duplicate issues on the issue tracker

today it's friday, usually friday is a working day in almost every country in the world. 
You are very lucky if you are with your family on friday 16:20 CEST, i'm not that lucky, i'm at work.
So, if you are with your family (or on vacation), there is no need to be forced to close this issue as soon as possible. Stay with your family and close them when you back at work, like everybody

bubnikv, 2019-04-26:
> It's not my fault if you don't properly tag the issues. It's a 2 years
old issue, with no tag set and with no official updates. If you don't want
duplicate issues, please properly handle the existing one or publish a
roadmap or something similiar.

What?

> today it's friday, usually friday is a working day in almost every
country in the world.

> You are very lucky if you are with your family on friday 16:20 CEST, i'm
not that lucky, i'm at work.

I am still at work, thank you very much.

> So, if you are with your family (or on vacation), there is no need to be
forced to close this issue as soon as possible. Stay with your family and
close them when you back at work, like everybody

If I am solving duplicates for you, I don't have time for improving Slic3r.
It is so simple.

vojtechkral, 2019-04-26:
> It's not my fault if you don't properly tag the issues. It's a 2 years old issue, with no tag set and with no official updates. If you don't want duplicate issues, please properly handle the existing one or publish a roadmap or something similiar.

The original issue can be easily found by typing _"adaptive slicing"_ into the search field. Doing at least a quick search before filing an issue is something that will honestly help both you and us. The original issue is still open, feel free to comment therein.

Please note that there's a great number of issues/requests/ideas etc. that could (and in many cases, should) be resolved/implemented, but it's next to impossible for the dev team to satisfy all requests, especially while also trying to release new versions on time.

Thank you for understanding.