prusa3d/PrusaSlicer #2175: Wrong layer count

guestisp, 2019-04-26:
### Version
1.42.0-beta2

The layer preview is showing one additional layer, all layers are "shifted" by one, in example, the layer1 doesn't show anything on the plate, the first "print" is at layer 2.

Slic3r is counting 458 layers but only 457 exists (as I can see in octoprint)
[image]
[image]

[amazing_bombul-albar_1.zip](https://github.com/prusa3d/Slic3r/files/3123223/amazing_bombul-albar_1.zip)

bubnikv, 2019-05-06:
It only happens if you visualize the travel paths.
@enricoturri1966 do you have an explanation for that?

guestisp, 2019-05-06:
> It only happens if you visualize the travel paths.

even if you choose "Retractions" or "Unretractions", just seen this right now after your comment

guestisp, 2019-05-06:
If useful: i'm slicing a test cylinder. Total layer count: 100, height 20.04 (why .04? The cylinder is 20.00)
When showing travels: new layer count: 103 height 20.54
When showing retractions: layer count: 101, height: 20.04
When showing unretractions: layer count: 101, height: 20.04

Travel has "priority" over the other two. Showing travel and retractions or any other combination, always show the bad number as with "Travel" only

enricoturri1966, 2019-05-06:
@guestisp

The extra layers shown by Slic3r, when travel paths, retractions or unretractions are selected, are there because those entities may start/finish on Z values which are different from those of the toolpaths and those Z values are used by Slic3r to show/hide them in dependence of the values set into the vertical slider. 
We can certainly improve this, but won't be a trivial change.

guestisp, 2019-05-06:
Understood, but this makes the preview not reflecting the reality. The preview is supposed to show exactly what is being printed, currently, you are showing more layers than the real.

As a workaround, can't you easily block the layer count update when choosing "retractions", "unretractions" and "travels" ?

guestisp, 2019-05-07:
It's not only the layer count to be wrong. If you enable "travel, the first layer is empty in the preview. The first printed layer would be the number 2

"retraction" or "unretractions" doesn't "shift" the first layer

rtyr, 2021-01-28:
Seems to be already fixed. Closing.

closed 2021-01-28 by rtyr