prusa3d/PrusaSlicer #2176: please support tab key to jump value fields in all pages

akust, 2019-04-27:
### Version
1.42.0-beta2

### Operating system type + version
MacOS Sierra 10.12.6

### 3D printer brand / version + firmware version (if known)
i3 MK3

### Behavior
**_description_**
often I need to adapt more then a single value but several in a raw. Usually the TAB key is supported to jump to next value field and Shift+TAB key to jump a value field back. This is working in some pages (partly) but not all. I'd like to ask you to reexamine all fields and make sure jumping via tab works as handy helper for easy serial input of numbers.

_examples for missing TAB support_
page Print Settings/Speed, Printer Settings/Advanced/

_example for fully working TAB as expected_
Filament Settings/Filament/

_example for partly working TAB_
Printer Settings/Layers and Perimeters/

**_Steps needed to reproduce the problem_**
- open page Print Settings/Speed
- select first value field
- press TAB  (/ Shift+TAB) several times

**_Expected Results_**
- jump to next (/ former) value field and pre-select it's value

* _Actual Results_
- nothing happens

_Is this a new feature request?_
no, I regard it as a bug.

#### Project File (.3MF) where problem occurs
(I doubt any file helps describing the issue)

FidelCapo, 2019-05-06:
Thank you for report. I can reproduce this issue.

Created SPE-914

kssi, 2019-11-23:
Has any progress been made on this in 6 months?

rtyr, 2021-01-28:
Already fixed. Closing.

closed 2021-01-28 by rtyr