prusa3d/PrusaSlicer #2177: {Feature Request] Option for LEGO Compatible Purge Tower for MMU prints

MrFlippant, 2019-04-27:
Feature Request

For MMU prints, the Purge Tower cannot be eliminated. It can be minimized with wipe objects and wipe to infill, but the tower itself cannot be completely eliminated. However minimized, we'll always end up with a brick (of varying density) of waste material. What if this brick could be made useful, along with any waste objects used during the print?

I suggest an optional feature to configure the size of the Purge Tower according to the specifications of LEGO style bricks. The unit can be as small as a single stud tile piece. As long as the purge tower is a multiple of that size, and topped off with a stud(s) of the appropriate diameter and height, the Purge Tower would suddenly become a useful object for LEGO creations of all kinds. This unit will still allow the Purge Tower to grow or shrink according to the needs of the print, even when stepped as the number of colors reduces. They could then serve as pedestals, risers, columns, walls, etc. They wouldn't even need to have a compatible base to be useful, only the correct proportions, and correctly sized studs on the very top. A base that could attach to the top of other bricks would be icing on the cake.

While a properly sized Purge Tower on the Z axis, as well as the final layers to create the studs may add a small amount of filament above the completed MMU object, it would be a small price to pay for the Purge Tower to be a functional, useful object that kids (of all ages) would love.  Spray paint it and use it to hold up your Millenium Falcon!!

Come on, let's reduce waste even more by making the unavoidable waste USEFUL for SOMETHING!

Who's with me!!?? :)

lukasmatena, 2019-04-29:
One of the reasons you cannot get rid of the wipe tower completely are the ramming extrusions. They are timing critical, doing them slower then they were designed can result in a failed print. Also, the extrusions produced during the toolchange are not reliable and can result in unacceptable quality issues on the resulting print. I have seen this when implementing the wipe to infill/object feature and this was the reason we added the filament parameter "minimal purge on the wipe tower".

Using the wipe into infill/object reduces waste by huge amount already. You can add as many lego bricks or whatever objects and use them for wiping - if you add enough, the wipe tower reduces to almost the ramming extrusions only. I don't thing it is worth it to try to reduce the waste a bit more and risk compromising reliability while doing so - a failed print wastes more plastic than anything and comes with an angry user as a bonus.

Not to mention that the implementation would require to pretty much write the gcode generation routines from scratch. That would require considerable effort during both developing and testing, which can be used more effectively on other awesome features.

MrFlippant, 2019-04-29:
Fully aware of the reasons the wipe tower cannot be eliminated, and the options for minimizing it. The fact that it can't be eliminated is why I put this suggestion up.

I do get that the tower is not made to be structural, and so can often be quite weak and brittle, especially when used for only ramming and a little priming, but it still usually makes a single wall perimeter and some interior supports for pretty much every layer. For a LEGO tower, it only needs resistance to crushing under load, and any LEGO that you put on top would likely not be heavy enough to crush it. If it were, you'd have more than one column holding it.

I definitely get that the generation of the tower would need to be modified, but I have to assume that there is already some pretty extensive calculating going on to determine its size and shape based on purge requirements and the above mentioned options for minimizing it. That calculation includes something that limits it in some way, and allows it to expand in other ways within a set of parameters. It might not be much harder than changing some parameters so that the limitations are within the specifications of LEGO bricks. Then, at the top, add on the necessary layers to add LEGO studs. The tower doesn't even really need to be a multiple of LEGO brick height, though that would be ideal.

But, yeah, I get that there are better things to do than try to reduce waste. It won't change the downward spiral our environment is in, anyway.

ZeGuigui, 2019-06-03:
Please note you can already set one of the dimensions of the purge tower in the settings (but not the other one so back to start!)

gaellafond, 2019-06-23:
Would that violate LEGO copyrights? I'm aware there is a lot of "fake" LEGO bricks out there, but that doesn't mean it's legal to do so.
You could print a LEGO purge block bottom and top, and glue them to your purge block to make a LEGO brick out of it. Not as nice, but that's a way to recycle them.

MrFlippant, 2019-06-23:
It is legal. The copyright/patent on the design expired years ago. That's why there are now so many "LEGO Compatible" brick toys on the market. They would not be on every toy shelf in the country if it were not legal to have off-brand bricks that are compatible. You may recall a time when all the competing brick toys were NOT compatible. They were always just a little bit off. Once the patent expired, every other brick toy maker changed their specs to make them compatible with LEGO.

Your idea of gluing a printed LEGO top/bottom to a purge tower isn't without merit. There are also a lot of parametric LEGO bricks on thingiverse to make it easy for someone to make their own bricks for purge objects. I was thinking of making a series of taller and taller bricks specifically for "secondary" purge  objects.

That said, I think think this is a worthwhile idea... but admittedly low on the priority list. For now, I'm experimenting with other ways of reducing waste.

MrFlippant, 2021-06-16:
Since you asked.... no. That's why I made this suggestion in the first place. If you read my posts, you'll see that I'm well aware of all the current options, and how the process currently works. But, I'm not deaf to the lack of support for this idea. I will now click that "Close with comment" button and hope that this is the last post in this issue.

closed 2021-06-16 by MrFlippant