prusa3d/PrusaSlicer #2178: Failure to load STL

Athrun80, 2019-04-27:
### Version
1.41.2 + win64

### Operating system type + version
Windows 10 Pro 64bit

Slic3r Prusa Edition
Version: 1.41.2+win64
Build:   Slic3rPE-1.41.2+win64-full-201811221508
Operating System:    MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: 
Microsoft Windows [Version 10.0.17763.437]

Summary of my perl5 (revision 5 version 24 subversion 0) configuration:
   
  Platform:
    osname=MSWin32, osvers=10.0, archname=MSWin32-x64-multi-thread
    uname=''
    config_args='undef'
    hint=recommended, useposix=true, d_sigaction=undef
    useithreads=define, usemultiplicity=define
    use64bitint=define, use64bitall=undef, uselongdouble=undef
    usemymalloc=n, bincompat5005=undef
  Compiler:
    cc='cl', ccflags ='-nologo -GF -W3 -O1 -MD -Zi -DNDEBUG -GL -fp:precise -DWIN32 -D_CONSOLE -DNO_STRICT -DWIN64 -DCONSERVATIVE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_NONSTDC_NO_DEPRECATE  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DWIN32_NO_REGISTRY',
    optimize='-O1 -MD -Zi -DNDEBUG -GL -fp:precise',
    cppflags='-DWIN32'
    ccversion='18.00.40629', gccversion='', gccosandvers=''
    intsize=4, longsize=4, ptrsize=8, doublesize=8, byteorder=12345678, doublekind=3
    d_longlong=undef, longlongsize=8, d_longdbl=define, longdblsize=8, longdblkind=0
    ivtype='__int64', ivsize=8, nvtype='double', nvsize=8, Off_t='__int64', lseeksize=8
    alignbytes=8, prototype=define
  Linker and Libraries:
    ld='link', ldflags ='-nologo -nodefaultlib -debug -opt:ref,icf -ltcg  -libpath:"c:\wperl64\lib\CORE"  -machine:AMD64 -subsystem:console,"5.02"'
    libpth=\lib
    libs=oldnames.lib kernel32.lib user32.lib gdi32.lib winspool.lib  comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib  netapi32.lib uuid.lib ws2_32.lib mpr.lib winmm.lib  version.lib odbc32.lib odbccp32.lib comctl32.lib msvcrt.lib
    perllibs=oldnames.lib kernel32.lib user32.lib gdi32.lib winspool.lib  comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib  netapi32.lib uuid.lib ws2_32.lib mpr.lib winmm.lib  version.lib odbc32.lib odbccp32.lib comctl32.lib msvcrt.lib
    libc=msvcrt.lib, so=dll, useshrplib=true, libperl=perl524.lib
    gnulibc_version=''
  Dynamic Linking:
    dlsrc=dl_win32.xs, dlext=dll, d_dlsymun=undef, ccdlflags=' '
    cccdlflags=' ', lddlflags='-dll -nologo -nodefaultlib -debug -opt:ref,icf -ltcg  -libpath:"c:\wperl64\lib\CORE"  -machine:AMD64 -subsystem:console,"5.02"'

  @INC:
    C:\PROGRA~1\Slic3r\local-lib\lib\perl5\5.24.0\MSWin32-x64-multi-thread
    C:\PROGRA~1\Slic3r\local-lib\lib\perl5\5.24.0
    C:\PROGRA~1\Slic3r\local-lib\lib\perl5\MSWin32-x64-multi-thread
    C:\PROGRA~1\Slic3r\local-lib\lib\perl5
    C:/PROGRA~1/Slic3r/lib
    C:/Program Files/Slic3r/lib
    .
OpenGL installation
GL version:   4.5.13492 Compatibility Profile Context 22.19.677.257
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon R9 200 / HD 7900 Series
GLSL version: 4.50
Installed extensions:
GL_AMDX_debug_output
GL_AMD_blend_minmax_factor
GL_AMD_conservative_depth
GL_AMD_debug_output
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_framebuffer_sample_positions
GL_AMD_gcn_shader
GL_AMD_gpu_shader_int64
GL_AMD_interleaved_elements
GL_AMD_multi_draw_indirect
GL_AMD_name_gen_delete
[... 265 lines cut ...]
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_OES_EGL_image
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)
custom built delta running duet wifi reprap version 1.21

### Behavior
Failure to load any stl file that has more than one object in it. If the objects are connected it loads just fine as it sees it as one piece, but multiple pieces separated from each other fails to load.

I've tried the newest version of slic3r PE and the issue remains. Cura and an older version of regular slic3r loads the STL without issue. Netfabb and Meshmixer also load the files just fine. This issue happens with models I design and export as well as any file downloaded from the internet.

 
[image]
[image]
[image]
[image]
[image]

bubnikv, 2019-05-04:
please attach the stl file.

bubnikv, 2019-08-01:
No model provided, closing.

closed 2019-08-01 by bubnikv