prusa3d/PrusaSlicer #2179: The application closes after generating gcode with wipe tower.

satiromarra, 2019-04-27:
1.42.0 beta 2 MacOS 10.13.6

The program closes at the end of the lamination with wipe tower.
I have tried with different files and options and always the same result.

Print settings > Multiple extruders > Wipe tower 
- Enable = enabled
- width = 25

Printer Settings > General > Advanced 
Use relative E distances = enabled

[image]

Check with this file:
[3DBenchy_-_Dualprint.3mf.zip](https://github.com/prusa3d/Slic3r/files/3124318/3DBenchy_-_Dualprint.3mf.zip)

lukasmatena, 2019-04-28:
@satiromarra Could you please explain what are you trying to achieve with the custom gcode magic? I'm not sure you know that the wipe tower ignores the toolchange custom gcode (#1245), so whatever it is, it will probably not work.

Anyway, I reproduced the situation (beta2, Ubuntu 16.04). The problem is indeed the `T[current_extruder]` in End G-Code. The last layer of the wipe tower unloads the filament and current_extruder ends up to be -1 at that point. Placing `T-1` into gcode then confuses the GCode preview and makes it access unrelated memory at line `GLVolume* volume = new GLVolume(tool_colors.data() + tool.value * 4);` in GLCanvas3D::_travel_paths_by_tool(...), hence the crash. I am not sure what the correct behaviour is, though. It of course should not just crash.

satiromarra, 2019-04-28:
Hi @lukasmatena 
Putting "T[current_extruder]" in final gcode is my fault since it is not necessary. Anyway, the application should not close and you should ignore it or show an error message.
But it only happens when I use the wipe tower, in the same configuration but without the tower no crash.

lukasmatena, 2019-04-29:
The crash should be fixed by https://github.com/prusa3d/Slic3r/commit/3ec17fae67cf24d934062bb3f1d817fe89e419e8 which is in current master and will be in the next release. The `T-1` command is left in the gcode and it is up to the user to decide whether it makes any sense or not.
Closing.

closed 2019-04-29 by lukasmatena