prusa3d/PrusaSlicer #2182: Feature request : BRIM per part

bgiot, 2019-04-28:
### Version
1.42.0 BETA 2

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

ochm, 2019-04-28:
Yes, it should be one of modifier feature. Thanks developers to add it in some future versions. :)

bubnikv, 2019-05-04:
duplicate of https://github.com/prusa3d/Slic3r/issues/651

closed 2019-05-04 by bubnikv