prusa3d/PrusaSlicer #2183: [Feature Request] Max speed reduction % in Cooling

guestisp, 2019-04-28:
Currently, when auto cooling is enabled, we can set a minimum layer printing time. If real print time would be less than the minimum, speed is reduced to get up to the minimum printing time.

Speed reduction is up to `min_print_speed`.

This is great, but could also lead to too much slow down on external perimeters and most of the time, the speed different on external perimeters is visible on the surface (different vibrations, ghosting and so on). 

I propose to add a `max_speed_reduction_percent` value to se the maximum speed reduction, in %, from the "original" speed. Is something that Simplify3D already has.

The `max_speed_reduction_percent` should still obey `the min_print_speed` value and the "computed" speed should not go slower than `min_print_speed`

bubnikv, 2019-05-04:
Apart from some special cases, where you may have different external perimeter speed settings for different objects or their parts, how that would help you? IMHO setting the min_print_speed fulfills the same purpose (apart from the cases I mentioned).

guestisp, 2019-05-14:
If I have an external perimeter speed set to 40%, internal perimeters set to 60mm/s and an infill set to 70mm/s , the current slow down due to cooling setting will be fixed, regardless the part being printed (external, internal, infill, ...) to `min_print_speed`.

My proposal is to add a new field so that we can lower all speed by, in example, 20%, so speeds would be: external: 20%(40%), internal 20%(60mm/s), infill 20%(70mm/s)

closed 2026-09-29 by prusaslicer-bot[bot]