prusa3d/PrusaSlicer #2185: Opening stl with double click do not mantain focus on Slic3r

dst74, 2019-04-28:
### Version
Latest stable 1.41.3
Latest beta 1.42 beta2

### Operating system type + version
Win10 Pro 1803
### 3D printer brand / version + firmware version (if known)
Mk3 3.7.0

### Behavior
Opening an stl by double clicking it does not maintain the focus on Slic3r. Focus come back to explorer.

ochm, 2019-04-28:
Do you have correct assotiations of .stl extension to slicer.exe?

EXOgreen, 2019-04-28:
I have 2 up to date windows 10 computers side by side. One of them keeps focus on the file explorer and the other holds focus on slic3r, and I haven't had the time to look into why.

dst74, 2019-04-29:
ochm, if I corretly understand your question, yes, because it opens by double click and the displayed icon is the slic3r one.

ochm, 2019-04-29:
> https://github.com/prusa3d/Slic3r/issues/2185#issuecomment-487461495 | view it
> on GitHub ] , or [
> | mute the thread ] .

So I don't understand what is the problem? 

​​-- 
MO

bubnikv, 2019-05-04:
I think the "problem" is the following:

If you double click on an STL file and the STL file is associated with Slic3r, and you just wait patiently, a new instance of Slic3r will start and load the STL. This new instance will get a focus.

However, if you double click on an STL file and the STL file is associated with Slic3r, and you do not wait patiently, but click back on the exporer or any other application, Windows will understand, that you want to let the slic3r to be loaded in the background, and you will have to give it a focus yourself.

dst74, 2019-05-05:
I'm sorry but it isn't. 
1. Close all the instances of Explorer and Slic3r
2. Open Explorer on the concerned directory.
3. Double click the stl (or click+Enter)
4. Wait
5. Slic3r opens the file get the focus
6. Instantaneously explorer takes the focus back on the window used to select the stl.
Regards

bubnikv, 2020-03-20:
I canot reproduce with PrusaSlicer 2.2.0-rc5. Closing.

closed 2020-03-20 by bubnikv