prusa3d/PrusaSlicer #2187: Not a seam issue - but what is it?

BirkBinnard, 2019-04-28:
### Version
142.0.Beta2

### Operating system type + version
Win10/64 Home Ver. 1809 current build

### 3D printer brand
Atom2 + Marlin

### Behavior
This is a bowl shaped piece with oval cross sections. These ridges appear on both sides of the print on opposite sides at the narrow ends. 

[image]

It looks like there is a change in tolerance for generating the hotend GoTo points as it encounters a smaller radius of curvature. Is there some input parameter that can control this?

beeb, 2019-05-03:
These artifacts are common in Delta printers. I'm afraid they stem from the motors themselves and are not due to the slicer. Used to have a delta for a long time and I used to see these kind of marks, although they would be more like concentric ovals for a "bowl" shape where the radii of the ellipsis are not constant. How does it look if you slice with a different software?

BirkBinnard, 2019-05-03:
That is a very interesting comment beeb - thanks. My printer is a delta and I've been printing with it for more than 3 years. I've not seen the pattern I showed prior to the latest release of this software, and I've used 2 or 3 other slicers as well with no similar signs.  In fact I did this test to check my printer's stepper motors: [http://birkbinnard.com/blog/smoothing-stepper-motor-movement/](url).

beeb, 2019-05-03:
Alright, my bad then, if you can confirm that slicing the same object with the same settings on an earlier version of Slic3r does not produce this result, I guess we could confirm this as being a bug in the current version.

BirkBinnard, 2019-05-03:
Thanks beeb, I can do that - but it will be a day or 2 off because I've got a long running print on my printer now. For some reason I haven't yet figured out or identified 142.0.Beta2 drives my printer at much slower speeds than previous versions.

BirkBinnard, 2019-05-05:
I reverted to 141.3 and the printspeed slowdown is no longer there. I'll let you know about the wiggles after more of the print is completed later today.

beeb, 2019-05-06:
Interesting. Can you compare the two gcode files (one from 141.3 and one from 142.0) with something like https://www.diffchecker.com/ ? Anything standing out from the preamble or other parts?

BirkBinnard, 2019-05-06:
I'll have to re-install 142.0.Beta2 to do the comparison, but I can do that. In the meantime here's an interesting result I just noticed. It's from this print, sliced with 142.0.Beta2 : [https://www.thingiverse.com/thing:3594128](url)
The yellow/bottom starts out as an elliptical shape, but ends at the top as a circular shape. The red/top is always a circle.
[image]
 The wavy pattern on the yellow part extends from the very bottom to the very top, but it is nowhere to be found on the red part. I have no idea why this should be of course, but it is quite noticeable when you look closely.

BirkBinnard, 2019-05-06:
OK - I downloaded the Bottom file from the above Thingiverse posting,sliced it with both versions, and saved the Gcode files. Unfortunately DiffChecker failed when trying to compare the 2 GCode files I made. So I installed the desktop version and that also crashed. My guess is the size of the files (just under 40 MB) is the cause of that. I tried several alternatives to DiffChecker but they also refused files this large.

I tried Zipping the GCode files but those end up at about 13 MB each and this forum has a 10 MB limit.  So I can't upload them either. Sorry about that.

What I did was export the Config INI file from 141.3, slice the Bottom STL shown above, and save the GCode. Then I installed 142.3-Beta2, imported the INI file I previously created, sliced the STL, and saved it's GCode. They differ in size by 0.01 MB.

I loaded each GCode file into Notepad++ and noticed that there are big differences in the G1 values right from the start. Here are the first 54 lines from both:

; generated by Slic3r Prusa Edition 1.42.0-beta2+win64 on 2019-05-06 at 08:50:53

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.45mm

M107
M104 S200 ; set temperature
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle
; Filament gcode
M109 S200 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 Z0.300 F7800.000
G1 E-3.00000 F2400.00000
G92 E0
G1 X59.042 Y-0.998 F7800.000
G1 E3.00000 F2400.00000
G1 F2160
G1 X59.059 Y0.002 E3.04810
G1 X58.999 Y3.593 E3.22081
G1 X58.812 Y7.241 E3.39651
G1 X58.351 Y12.234 E3.63769
G1 X57.861 Y15.888 E3.81499
G1 X57.254 Y19.445 E3.98856
G1 X56.522 Y22.980 E4.16219
G1 X55.660 Y26.498 E4.33637
G1 X54.654 Y30.032 E4.51313
G1 X53.531 Y33.471 E4.68711
G1 X52.240 Y36.959 E4.86599
G1 X50.882 Y40.218 E5.03582
G1 X48.732 Y44.745 E5.27684
G1 X46.993 Y47.971 E5.45313
G1 X45.223 Y50.933 E5.61909
G1 X43.328 Y53.805 E5.78456
G1 X41.430 Y56.419 E5.93992
G1 X39.198 Y59.208 E6.11174
G1 X36.964 Y61.727 E6.27367
G1 X34.616 Y64.115 E6.43476
G1 X32.292 Y66.246 E6.58642
G1 X29.614 Y68.446 E6.75310
G1 X27.055 Y70.309 E6.90535
G1 X25.411 Y71.396 E7.00015
G1 X22.573 Y73.077 E7.15879
G1 X19.671 Y74.557 E7.31546
G1 X16.737 Y75.822 E7.46913

; generated by Slic3r Prusa Edition 1.41.3+win64 on 2019-05-06 at 08:49:46

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.45mm

M107
M104 S200 ; set temperature
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle
; Filament gcode
M109 S200 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 Z0.300 F7800.000
G1 E-3.00000 F2400.00000
G92 E0
G1 X59.042 Y0.996 F7800.000
G1 E3.00000 F2400.00000
G1 F2160
G1 X58.999 Y3.593 E3.12489
G1 X58.812 Y7.241 E3.30059
G1 X58.351 Y12.234 E3.54177
G1 X57.861 Y15.888 E3.71907
G1 X57.254 Y19.445 E3.89264
G1 X56.523 Y22.975 E4.06601
G1 X55.660 Y26.498 E4.24045
G1 X54.654 Y30.032 E4.41721
G1 X53.531 Y33.471 E4.59120
G1 X52.240 Y36.959 E4.77007
G1 X50.882 Y40.218 E4.93990
G1 X48.732 Y44.745 E5.18092
G1 X46.996 Y47.966 E5.35695
G1 X45.223 Y50.933 E5.52317
G1 X43.328 Y53.805 E5.68864
G1 X41.430 Y56.419 E5.84400
G1 X39.198 Y59.208 E6.01582
G1 X36.964 Y61.727 E6.17775
G1 X34.616 Y64.115 E6.33884
G1 X32.292 Y66.246 E6.49051
G1 X29.614 Y68.446 E6.65718
G1 X27.055 Y70.309 E6.80944
G1 X25.411 Y71.396 E6.90423
G1 X22.573 Y73.077 E7.06287
G1 X19.671 Y74.557 E7.21954
G1 X16.737 Y75.822 E7.37321
G1 X13.774 Y76.876 E7.52449

Needless to say I have no idea what to make of this. My sense is that some internal parameter or setting about how individual Goto points (G1 values) get calculated changed between 141.3 and 142. I have some understanding about how these values are calculated from old experience with generating control files for large milling machines, but that was a long time ago.

beeb, 2019-05-06:
Ok thanks for your work. From what I see of your two gcode files, they are essentially identical. The extruder positions (`Ex.xxx`) values are a bit different because there is an additional instruction in the 1.42.0 version (`G1 X59.059 Y0.002 E3.04810`) that is not found in the other one. That explains the shift in E values but for the most part, the X and Y coordinates are identical, so are the feedrates (speeds). So these 2 files should print at the same speed.

That's how far I can compare them obviously without having the full files, but I don't see anything strange. Maybe someone more experienced can spot a big difference that I don't see.

Is it possible to host the files somewhere else so that I can download them?

beeb, 2019-05-06:
Regarding the wavy pattern, if you're saying that the files generated in 1.41 print noticeably faster than the files from 1.42, this could explain the difference in surface aspect. If the print head moves faster, any microstepping artifact is spread over a bigger distance and does not show up as much. Finding a way to make the files from 1.42 go the same speed as the files from 1.41 (which should already be the case if I look at the gcode you posted above) could well solve your issue.

BirkBinnard, 2019-05-06:
I think you may have identified the issue - it could very easily be the difference in print speed. 

My"normal" print speed is 80 mm.sec. When I switched to 142.3 I was surprised that my printer was running significantly slower. My guess is around 20 mm/sec. When I saw this I thought I could just speed it up by twisting the knob on my LCD panel, but that had no effect.  So I just let the print finish at the slower speed. I was printing the screw-top bowl I referenced in the Thingiverse link above. As I recall the print took almost 2 days to finish.

When I switched back to 141 i found that my printer printed at the "normal" speed I was used to, and also that the LCD panel knob was able to control the printer's speed just as it had in the past. I've just restarted a print now with 141 and my guess is it will not have the wavy pattern. The print should finish sometime tomorrow so I'll let you know then.

But...what could be the cause of the behavior change between 141 & 142? My printer's firmware has not changed - it is (as far as I can tell) standard Marlin with tweaks to define my delta printer's specific geometry. The firmware is literally several years old and 100% stable as far as I can tell. The print speed difference was very obvious, as was the loss of any effect from the LCD knob.

beeb, 2019-05-06:
It would help if you could provide a full gcode file that prints at the normal speed, and a full gcode file that you know for sure prints slower with the same settings. From what I saw in the gcode you posted earlier, there is no reason this should happen. This leads me to believe some settings were different in the initial 1.42 file that printed slower.

Besides that, you could also try to artificially slow down the print with a 1.41 file and check for wavy patterns. No need to print the whole piece, 10 or 20mm high should suffice to see if the artifacts are present.

BirkBinnard, 2019-05-06:
Here you go: 

https://drive.google.com/file/d/16pSvkiRecRSJMSTnM_OOnfj81XYvYSR_/view?usp=sharing

This is a 30.6 MB ZIP file with both GCodes, the INI file I used for both slicers, and the STL file itself in case you want to try that.

beeb, 2019-05-06:
Thanks. Indeed the files look very similar. I don't understand how one can print faster than the other, except for an obscure bug in your printer's firmware. If you are sure that this most recent 1.42 file prints differently (not sure you have verified this since making sure the configs were identical by exporting the .ini), then that's as far as my knowledge can help you..

BirkBinnard, 2019-05-06:
I agree - most mysterious. Since I have both versions on my system now I'll be able to do more testing with a smaller STL file after my current print is finished.I'll post again in a new thread if I come up with anything that makes sense. Feel free to close this one if you like.

rtyr, 2021-01-28:
I believe this issue is no longer relevant and was probably hw related. It is possible there was some speed difference between those two PS versions because of the changes in cooling logic (slowdown, minimal layer time). 

Closing.

closed 2021-01-28 by rtyr