prusa3d/PrusaSlicer #2188: Nozzle crashes in the print on first layers

Rarstorm, 2019-04-29:
Slic3r Prusa Edition 1.42.0-beta2

Windows 10 Home Single Language
Version 1809

During print. Always in the first 2 or 3 layers, the nozzle stops and crashes into the print. It did not happen on the Alphas. I'm 100% sure is the slicer, because if I go back to the lates alpha, or slice with other program like Cura, there is no problem with the print.

Printer 
Anycubic i3 Mega
Board Trigorilla
Marlin 2.0

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

bubnikv, 2019-04-29:
This is interesting. Do you have an idea why this is happening? Did you
compare the G-codes generated by 1.41.2 vs. 1.42.0-beta2?

bubnikv, 2020-03-26:
No response, closing.

closed 2020-03-26 by bubnikv