prusa3d/PrusaSlicer #2193: Skirt loops not being honored, only producing single loop

skellatore, 2019-04-30:
### Version
1.42.0-beta2+win64+full-201904140830

### Operating system type + version
Windows 10 64bit

### 3D printer brand / version + firmware version (if known)
Prusa Original MK3s

### Behavior
* _Describe the problem_
When slicing multi-coloured thing (could happen on a single-coloured thing but not tested) with a skirt loop count of 5, at the output stage the skirt is reduced to a single loop.

* _Steps needed to reproduce the problem_
1) Load a multi-coloured model

2) Set the extruders

3) Set the skirt settings to:
loops: 2+
distance: 5mm
height: 1 layer
Minimal extrusion length: 0

4) Slice

5) Notice how during slicing the preview window shows the correct amount of loops, once it's finished the loops shown is reduced to 1. 

6) export the GCODE and check - you'll see its a problem with both visualisation and GCODE generation.

Preview during slicing:
[image]

Preview after slicing:
[image]

#### Project File (.3MF) where problem occurs
https://www.dropbox.com/s/kow50vpm7i1n2io/Ko_Color%20v1.3mf?dl=0
sorry - was too big to upload here.

bubnikv, 2019-05-04:
While you indicated that you are using the I3 MK3 printer, your printer profiles are heavily modified from ours. I wonder what printer are you printing on?

I would have to look into the code, but it is quite possible that the number of skirt lines is not respected in the single extruder multi-material mode. The skirt lines are usually used to purge or pressurize the extruder, and to verify that the bed is well leveled around the future print. As the purge and pressurization is taken care of by the wipe tower of the single extruder multi-material configuration, there is no need to extrude more than a single loop.

It is different in the conventional multi extruder setup, where one wants to extrude a single loop per extruder to pressurize all the printing extruders before the print. Slic3r lacks here, as it does not currently support a wipe tower for the conventional multi-extruder setup.

skellatore, 2019-05-04:
i’m using a Prusa MK3s with a post processing script and a Palette2..

A regardless of the reasons you stated, if I tell it 4skirts, I expect 4 skirts! works fine in stable.

lukasmatena, 2019-08-26:
I managed to reproduce the bug. It is in there for a long time, Slic3rPE 1.41.2 and PrusaSlicer2.1.0-beta2 are affected in the same way. The bug is not completely harmless, it can result in printing the higher skirt lines in air. Enclosing a 3MF to reproduce, the original posted by @skellatore is no longer available.

Added to my todo list.

lukasmatena, 2020-01-14:
I apparently forgot to attach the 3MF, so here it is. I see two things that could be better:

1. Part of the skirt prints in air on second layer

2. It might be useful to use the skirt for priming after extruder switching (especially for multiextruder printers with no wipe tower). It might therefore make sense to print the individual print loops with different extruders, just before objects are printed.

[skirt_bug.zip](https://github.com/prusa3d/PrusaSlicer/files/4058340/skirt_bug.zip)

bubnikv, 2020-01-14:
The skirt generator used to prime all printing extruders at the 1st layer if enough skirt lines were configured, while at the other layers the skirt used to be printed with the 1st extruder printing the layer.

There was a bug introduced quite a long time ago, where not enough skirt lines were extruded if some extruders were not needed to print the 1st layer.

Newly the skirt generator works the same way on all layers: All the extruders planned to print a layer are primed over the skirt if enough skirt lines are configured.

Fixed with 10d1b459bb305683a6c919ced40f98d1d6553928. The fix will be part of the next PrusaSlicer 2.2.0 alpha. Closing.

closed 2020-01-14 by bubnikv