prusa3d/PrusaSlicer #2194: thin walls: part of object missing

guestisp, 2019-04-30:
### Version
1.42.0-beta2+linux64

When slicing the attached STL (a filament test), with a 0.4 nozzle and thin wall enabled, some part of object are properly rendered, but other part are missing. Disabling thin walls does the oppositve: what was printable thank to the thin wall, will be removed, and what wasn't printable before, became printable.

I would expect that thin walls will add printable parts to the sliced code, and not to reduction.
In example, look at the 0.4 on the bottom. With thin wall, the lower part of number "4" is missing, without thin walls, that part is printable but we loose something else

[image]
[image]

[Configurable_Filament_Swatch_VQ.zip](https://github.com/prusa3d/Slic3r/files/3133165/Configurable_Filament_Swatch_VQ.zip)

guestisp, 2019-05-04:
I've also seen another strange behaviour here. Look at the top solid infill. With thin wall disabled, the top infill will be constant and uniform, with the same patter, producing a better surface finish. Enabling the thin wall, it make a single straight line. This isn't good in top solid infill as you'll get different finishing in the top.

Between the two, based on these images, the top solid infill without thin wall is better.

shawnvanlan, 2019-05-11:
I have the same problem even when I redo design and make all walls at least .55mm thick.

closed 2026-09-29 by prusaslicer-bot[bot]