prusa3d/PrusaSlicer #2195: Does not fill on several thicknesses.

CHATON89, 2019-05-02:
### Version 1.42 beta
windows 8.1

### 3D printer brand / ENDER 3

### Behavior:
The slicer leaves holes on several layer thickness while under CURA there is no problem.

_Is this a new feature request? _

#### Project FilE: ### Version 1.42 beta
windows 8.1

### 3D printer brand / ENDER 3

### Behavior:
The slicer leaves holes on several layer thickness while under CURA there is no problem.

_Is this a new feature request? _

#### Project FilE: https://www.thingiverse.com/thing:2030296/files

[image]

jmd73, 2019-05-02:
There are errors in the model which Slic3r does not handle. If you repair the model with the Netfabb service or other program, it will slice correctly.
Since you seem to be on windows 10, you can right click on the exclamation mark to the left of the model name on the right side of the screen it will be repaired by Netfabb.

CHATON89, 2019-05-02:
I am working on windows 8.1.

CHATON89, 2019-05-02:
I iron the file under solid work there is more error my problem is still present.
[image]

jmd73, 2019-05-02:
[Black_Goku-repaired.zip](https://github.com/prusa3d/Slic3r/files/3138591/Black_Goku-repaired.zip)
Sorry I missed that you were on Windows 8.
I repaired it in Slic3r and exported it again as STL.
Please try it and see if it works.

CHATON89, 2019-05-02:
The new file works well thank you, how did you do to fix it?

jmd73, 2019-05-04:
I repaired it as I described in my first reply.

jmd73, 2019-05-05:
Took another fast look at the object and it seems to be made up of lots of parts. If you split it to objects in Slic3r you get several parts, and Slic3r does not report any errors on the separate parts.

closed 2026-09-29 by prusaslicer-bot[bot]