prusa3d/PrusaSlicer #2196: Settings UI non responsive when editing if Plater has large STL

jimmy00784, 2019-05-02:
### Version
1.42 beta2

### Operating system type + version
Arch Linux 64 bit (5.0.9 kernel)

### 3D printer brand / version + firmware version (if known)
CoreXY printer with Marlin 1.1.9

### Behavior
When background processing is enabled, if a large STL is already on the bed, changing in Print and Printer settings, changing values that directly effect the GCode generation results in GCode refresh immediately, however, causing the UI to become non-responsive when the new GCode is being generated.

Steps to recreate:
1. Add a large STL or scale an stl to near max volume.The issue is easier to observe that way.
2. Switch to Print Settings -> Layers and perimeters
3. Change Layer height, perimeters, shells etc

Ideally, background processing shouldn't interfere with UI responsiveness.
At the least, background processing should be disabled in Settings views so the user experience isn't adversely impacted.
I can live with background processing kicking back in as soon as I switch to Plater.

bubnikv, 2019-05-04:
The background processing shall be terminated at regular check points sprinkled around the background processing code. We need to verify that there are enough of cancellation points in the G-code generator.

bubnikv, 2019-08-01:
Please provide a 3mf for which you have the issue.

bubnikv, 2019-08-01:
Also please provide the content of the status line - the name of the slicing step, which could not be easily canceled.

jimmy00784, 2019-08-14:
This seems to be fixed in PrusaSlicer 2.0.0
Background processing is minimally affected by changing settings while processing is not complete.

closed 2019-08-14 by jimmy00784