prusa3d/PrusaSlicer #2199: Set retraction speed not respected during wipe movement

ogland, 2019-05-03:
### Version
1.42.0 beta1 win64 full

I had some issues with filament missing after deretraction. Turns out using the option "wipe while retracting" sets a mysterious feedrate of 96 mm/s for all axis during wipe movement.

I suggest fixing this, as ripping the bead isnt good for messy filaments like PETG, where the option is usefull but breaks from this.

Here's what 0.6 mm retraction with wipe while retraction looks like. Note that only 0.03 mm of that retraction actually respects the set speed, which is useless.

```
G1 F5760
G1 X129.635 Y110.199 E-0.57000
G1 E-0.03000 F1800.00000
```

And the same without wipe looks normal and respects the set retraction speed of 30 mm/s

`G1 E-0.60000 F1800.00000`

Sebastianv650, 2019-05-03:
Read #2109 and you should understand why there is not the retraction speed you think should be there..

ogland, 2019-05-03:
I read it and need to calculate the feedrate differently I guess.

bubnikv, 2019-05-04:
is this issue solved for you? Would you please close it?

closed 2019-05-04 by ogland