prusa3d/PrusaSlicer #2202: Bad seams on non-straight surfaces

guestisp, 2019-05-04:
### Version
all versions

I'm fighting with some strange seam issues.
Every time a seam is placed on a non flat non straight surface , some blobs and holes are left on the surface.

The attached photo are showting two prints, with the same exact settings (just printed the cylinder first, then removed the part from the plate, added the cube, sliced and printed, nothing more). The cylinder has a very rough surface, with some holes (like underextrusion, but i'm not underextruding) and full of blobs, exactly where the seam is supposed to be (in a cylinder, there isn't any corner to hide it). The cube is almost perfect, the seam is hidden in the corner and the surface is flat.

I've also attached some thin wall test, printed in vase mode, with no seam, the are al perfect, surface smooth, flat, no blobs, zists or similiar.

guestisp, 2019-05-04:
[image]
[image]
[image]
[image]
[image]
[image]

guestisp, 2019-05-04:
current settings:
retract_length=3
retract_lift=0
retract_speed=50
retract_layer_change=1
wipe=1

with or without wipe doesn't chage anything, same behaviour.
I've tried increasing and lowering the retraction, same behaviour

tried on multiple printers, so it's not an hardware issue.

Identical settings (speeds, retractions, extrusion width, extrusion multiplier, ...) copied and pasted from Slic3r PE to a different slicer, doesn't show the issue.

I think there is something strange in how Slic3r make the seam. Tried with nearest and aligned, same results. With random it's even worse. With rear is perfect, the seam is located in the rear, very small, no holes, no blobs, but obviously there is a huge line in the rear of the piece.

So, as long rear and nearest/aligned make the same seams, but just alined differently, I'm thinking something strange is happening during the seam positioning by slic3r.

With an hardware issue (leaky extruder or similiar), even the rear seam would be broken as the retraction/wipe movement would be the same, just in a different position. Even a cube would be bad with an hardware issue.

guestisp, 2019-05-04:
This is the same cylinder, with the same settings as before, but with seam aligned on the rear. As you can see the seam is perfect, no blobs , holes or scratches.

I would expect the same even with seam aligned or nearest, as the hardware is the same.
[image]

guestisp, 2019-05-16:
I'm still doing tests about these seams.
I would expect that all kind of seams would be printed in the same way, only the seam position should change.

That's not the case.
- A rear seam woul make a huge cut, all along the printed part (as shown in the photo)
- A random seam would make very,very small blobs/zists on the surface (not a cut, just a single blobs)
- A nearest/aligned seam will act even more differenty producing huge seam.

I think there is a bug somewhere, the seam position should not affect the way the seam is printed.
What I would expect, as this setting is called `seam_position` and not `seam_type` is the same blobs/zists as the random one, but aligned, neared or on the rear.

Actually, we have at least 3 kind of different seams.

guestisp, 2019-06-07:
Any advice on this ?

The only good seams are `random` or `rear`. Random is adding tons of blobs (as it should be) but very very small. Rear add a perfect seam on the rear. The other two are making the whole surface very bad, huge blobs like a lot of overextrusion.

guestisp, 2019-06-07:
What I saw is that the seam is not placed on the exact corect but a little bit after or a little bit before the corner it self. Even by using the rear seam, slicer is trying to place the seam on a rear corner, if any, it does it well, but not perfectly. Most of the time the seam is shifted from the corner itself.

Look at this image. There is a corner available (in reality, there are two), but the seam, sometime, is placed off the corner, before or after and not over the corner as it should be.

[image]

brew99, 2019-11-22:
I'm finding the same issues as you on the curved items without a defined corner.   Using Prusaslicer 2.1

Anyone have suggestions?

closed 2026-09-29 by prusaslicer-bot[bot]