prusa3d/PrusaSlicer #2203: What happened to the Print Button?

frakman1, 2019-05-05:
### Version
Slic3r 1.42.0-beta2 - Prusa Edition (win64)

### Operating system type + version
Win 7 64bit

### 3D printer brand / version + firmware version (if known)
Prusa MK3S
Firmware: 3.7.0-2201

### Behavior
What happened to the `Print` Button?
Disable USB/serial connection
Under Configuration->Preferences, the following check mark item is no longer there:

[image]

Therefore, I can't connect to my printer and print directly like I could with the standard release of Slic3r PE 1.41.3+win64

Is there a different way to print directly to my printer?

uniacke1, 2019-05-05:
This was a deliberate choice during the rewriting of the core components, from what I have read. It may be restored but for now your only option is a third party tool like pronterface or octoprint to print directly from PC.

shro70, 2019-05-14:
I need this print button. I like to USB USB printing for small print.

nate98252, 2019-05-17:
Kind of messed up to take away a useful feature and force people to be limited to one specific way of printing. I understand there are other programs that allow you to print via USB but if you want people to use your software above all others.....don't limit what they can do.

vojtechkral, 2019-05-17:
I believe this is a duplicate of #2046, closing in favor of that...

closed 2019-05-17 by vojtechkral