prusa3d/PrusaSlicer #2207: [Enhancement]: Progressive part cooling fan ramp up

beeb, 2019-05-05:
I sometimes get a "Bed preheat error" in the following style of scenario:

 * The heatbed temperature is relatively high (i.e. 85C)
 * The part cooling fan needs to be 100% (for best quality)
 * The printed part is small

This happens because the blower fan cools the bed so much that the PID loop needs to go crazy and triggers the error (in my understanding). It sometimes also happens with the nozzle, the PID loop cannot get it back up to temp fast enough and it triggers an error.

My proposed solution would be to do a progressive ramp up of the fan speed when it goes from a very low values (TBD) to a very high value (TBD). Either as a configuration option for the first few layers where this is the most likely to happen:
Instead of the current "Disable fan for the first ... layers" it would be "Ramp up fan over the first ... layers"
or as a global setting that would happen anytime the fan needs to increase dramatically.

Let me know what you think!

jokohoko, 2019-05-12:
My solution is to place such small parts on the side of the build bed. Since the sensor is mount in the center.

beeb, 2019-05-12:
This is more a workaround than a solution

ManuGithubSteam, 2019-08-18:
This will also help with some filaments who suffer from visual defects when the fan kicks in at 100% the first time. Some like it a bit slow at first to get used to it.

Nandox7, 2020-05-19:
Looking for this as well, either set values across different layers or have it increase by % per layer till a max value.
Is good to avoid big thermal differences for filaments that are more sensitive to it like ABS.

davidhbrown, 2020-11-24:
> My solution is to place such small parts on the side of the build bed. Since the sensor is mount in the center.

That works for a bed thermal overrun, but I've been having this problem with the hot-end itself on a 2.5S with recent firmware. Can't move the part away from the nozzle! :-)

bubnikv, 2020-11-25:
> That works for a bed thermal overrun, but I've been having this problem
with the hot-end itself on a 2.5S with recent firmware. Can't move the part
away from the nozzle! :-)

Maybe you need to re-run the PID extruder calibration on your printer?

davidhbrown, 2020-11-30:
Thanks, bubnikv, but I've done that... with and without a silicone sock (which helps until it picks up crud and knocks your print over). There's more discussion at https://github.com/prusa3d/Prusa-Firmware/issues/1746 which, as you can infer, points the blame to the firmware because it didn't used to happen. Maybe the temperature limits are set thinking we all got the 24V heater (2.x are 12V). Anyway, this enhancement would be of greater utility than a firmware patch for a discontinued model, so I wanted to add my +1.

beeb, 2020-12-05:
Latest beta solves this I think! Will try it out

bubnikv, 2020-12-05:
Cool, closing.

closed 2020-12-05 by bubnikv