prusa3d/PrusaSlicer #2208: [Enhancement]: don't crash if bed size is too large

toasta, 2019-05-06:
### Version 
1.41.3+win64

### Operating system type + version
Win10

### Behavior

if you go to "Printer Settings" => "Bed Shape > Set"and set size to e.g.
2500x2100 (just times 10) it ponders for a bit and finally crashes
setting it to 10x10 meters should do the same.

dunno if it just ooms as it just closes.

_Is this a new feature request?_

kinda...

bubnikv, 2019-08-01:
Slicer should not crash, that is not a question.

Internally much of the calculation is done using 32bit fix point integers scaled to 1um, allowing for dimensions roughly -2 meters to +2 meters. Also there are some thresholds hard coded internally, which work well for nozzles in the range of 0.25 to 1mm. You are likely printing with some huge nozzle, therefore these internal thresholds are not suited to your printer. The best you can do is to work with coordinates in cm, not in mm, then everything will work.

YuSanka, 2020-04-01:
Close as a duplicate of #2877

closed 2020-04-01 by YuSanka