prusa3d/PrusaSlicer #2210: Need a way to specify tool number per extruder

DrLex0, 2019-05-06:
### Version
1.42.0-beta2

### Operating system type + version
Any

### 3D printer brand / version + firmware version (if known)
Any printer running Sailfish or MakerWare (and perhaps other firmwares)

### Behavior
* When creating a single-extruder printer profile for a multi-extruder printer running Sailfish or similar firmware, Slic3r always assumes that the profile is meant for T0, and will insert M104 commands for T0 even if the profile is meant for T1. This means I cannot make a single-extruder printer profile that prints with T1 only.
  Before the UI change introduced in 1.42, it was still possible to work around this by forcing the correct extruder number for the object in the settings dialog, but this has now been removed and the ability to change extruder number is only available if the printer profile has more than 1 extruder. I have to rely on a post-processing script that strips the T0 argument, which works by the grace of the GPX program keeping track of what extruder was last selected.
* _Steps needed to reproduce the problem_
  * Slice the file from the attached profile, which as can be seen from its start G-code is meant for T1.
  * Look at the generated G-code around the change to the second layer.
* _Expected Results_
  * There is some way in Slic3r to specify that the object is to be printed with T1 (or any other number), and any M104 commands inserted by Slic3r get a T1 argument.
* _Actual Results_
  * Any M104 commands inserted by Slic3r always have a T0 argument and there is no way to specify that we want to use T1. The wrong extruder heats up after layer 1 and the print turns into a mess.

_Is this a new feature request?_
Yes, although it is not really new. I already filed [this issue on mainline Slic3r](https://github.com/slic3r/Slic3r/issues/4003) but it has only been gathering dust ever since.

I don't know what is the best way to solve this. One possible way is to add a “Tool number” field to the Extruder settings in the Printer Settings. Obviously this must be an expert option. (Ideally it would not allow entering the same number for two different extruders in the same profile, but since it is an expert option anyway, that would only be icing on the cake.)

#### Project File (.3MF) where problem occurs
[SingleExtruderTool.3mf.zip](https://github.com/prusa3d/Slic3r/files/3150173/SingleExtruderTool.3mf.zip)

DrLex0, 2022-02-12:
No.

RocketSci420, 2022-02-15:
@lukasmatena I've actually tried both defining the printer with both extruders and the new gcode substitution. As @DrLex0 mentioned, setting up the machine with both extruders and only printing with the left can work, but it's not ideal. Another issue (possible bug) I found, you cannot add gcode modifications at a specific layer in the preview window with two extruders defined. I've tried in both 2.4.0 and 2.4.1 with the same result. I can define the gcode modifications at layers, ex. M104 S200 T1, but when you click "slice now" it deletes all of the custom gcode modifications and slices the model without the modifications. Again, this only happens when multiple extruders are active. 

I've also tried the gcode substitutions, but I am not very familiar with ECMAScript language. I successfully tested a find \M104 S.+T0 and replaced it with M104 S205 T1. But I would like to figure out next is how to get that substitution to work by changing only the T0 to a T1 while returning the M104 S. that it found. The S temp may be variable for this substitution case and I want it to return the same M104 S. The gcode substitution seems to address my issue, but I'd still prefer to have the ability to define a tool number per extruder. And I agree with DrLex that this gcode substitution feature belongs in the printer settings rather than the print settings.

closed 2026-09-29 by prusaslicer-bot[bot]