prusa3d/PrusaSlicer #2211: [Enhancement] Trick to help corners warp less

Stoddles, 2019-05-07:
### Version
Any version

### Behavior
Parts with sharp corners have a high chance to warp if they are printed in certain materials(ABS,Nylon,etc). An option that reduces the amount of material in sharp corners (e.g the inside perimeters or infill are rounded near the corner instead of following the outer contours continuously) will help parts to fight lifting from the print bed. 
* _Steps needed to reproduce the problem_
Make a part with a 90 degree bend and lots of perimeters out of a high temp material.
 
_Is this a new feature request?_
Yes

guestisp, 2019-05-07:
I think the original slic3r has a feature called brim ears that add a simple brim just around the corners

Stoddles, 2019-05-08:
Brim doesn't always work. I've been using a full brim with extra extrusion multiplier to make the brim stick to itself, with a heated enclosure on freshly cleaned buildtak. Printing as slow as I can with a fat first layer and squished into the bed. Pure polycarbonate is just really strong at pulling up sharp corners compared to petg and pla.

bubnikv, 2019-08-01:
You may add custom "micky ears" inside PrusaSlicer by adding low profile cylinders to the corners. For example, click on the object, select "add part->cylinder" from the pop-up menu. You have to be in an Expert mode.

rtyr, 2021-01-28:
Same as https://github.com/prusa3d/PrusaSlicer/issues/4568 and https://github.com/prusa3d/PrusaSlicer/issues/2954. I am closing this one.

closed 2021-01-28 by rtyr