prusa3d/PrusaSlicer #2212: drag and drop stl in to slicer

mitchy2233, 2019-05-07:
### Version
_Version of Slic3r Prusa Edition used goes here_ 1.41.3+win64

drag and drop stl file into slice seriously needs to be installed...im blown away that it isn't

bubnikv, 2019-05-07:
it always worked.

mweinelt, 2019-05-07:
Works for me with 1.41.3 on Linux.

mitchy2233, 2019-05-08:
Oddly it has to be a file from the desktop. Can’t be a folder located anywhere else. Makes for a extremely messy desktop. I’ll do a screen capture and show what I mean.  Maybe a bug ?

Creat, 2019-05-11:
also not true for me, I can drag&drop stl files from anywhere onto the plater and it works fine. I'm also sure of this as I add all files this way, and none are ever on he desktop.

You also haven't even mentioned what OS you're on (and which version).

mkoistinen, 2019-05-11:
FYI. It works for me on v1.42.0b2 on MacOS Sierra. Both dragging an STL into the program's icon or into the application window.

bubnikv, 2019-05-23:
The original report is obsoleted by the PrusaSlicer 2.0.0. Closing.

closed 2019-05-23 by bubnikv