prusa3d/PrusaSlicer #2214: Crash when deleting object if Layer Editing Toggled

maxcom99, 2019-05-07:
### Version
1.42.0-beta2+win64-full-201904140830

### Operating system type + version
Win 10 Pro x64 1809 build 17763.475

### 3D printer brand / version + firmware version (if known)
Prusa MK3S using 3.7 firmware

### Behavior
Two ways to reproduce, one fatal.
Non fatal -
Import ANY STL click "Layers Editing"
Delete object

Fatal-
Import ANY STL click "Layers Editing"
Change Layer heights
Delete object

Resolution:
Closing the layers editing UI allows the object to be deleted.
[image]
[image]

lukasmatena, 2019-05-07:
Thanks for reporting. I believe this was already fixed by https://github.com/prusa3d/Slic3r/commit/045879f68a07758aef43e4ae475d664f1c12f418, so it should be working in the next release.

bubnikv, 2019-05-08:
It will be fixed in the next release. Closing.

closed 2019-05-08 by bubnikv