prusa3d/PrusaSlicer #2216: [Feature Request] Filament Wear Prevention

rcmaniac25, 2019-05-10:
### Version
1.41.3+win64

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Prusa MK3 (stock R3)
Firmware: 3.7.1-RC1 (just upgraded as of this ticket creation)

### Situation
Some prints I've done have many areas where the filament is retracted, the extruder moves, extrudes a single dot of filament, then repeats retracting, moving the extruder, and making another dot. Repeat 3-60+ times. I had at least 1 print fail and, once the filament was removed, I saw the drive gears had completely chewed through the filament. My guess is because of the extreme retractions with minimal movement.

### Idea
A setting (or just changing the internals to do this by default) where small extrusions that are doing a single dot of game fill, or making a bunch of small stick-like towers for a model, etc. has to be broken up by longer extrusions for the sake of keeping the filament from getting chewed up. Perhaps a counter or minimal extrusion distance per number of gcodes before a longer extrusion needs to be done?

guestisp, 2019-05-10:
Can't you increase the minimum travel before retract to avoid this?

rcmaniac25, 2019-05-10:
I've not heard of that setting before, but I found it and read a bit. That setting is to prevent unnecessary retractions if not moving to a new location further then the threshold specified.

My idea wasn't for distance the extruder travels, but how much filament is extruded over the course of multiple retractions.

Extrude 0.2mm of filament, retract 0.8mm, undo, extrude 0.2mm, extrude 0.2mm of filament, retract 0.8mm, undo, repeat. Do that 5 times and you've extruded a total of 1mm of filament, but at least 1.6mm of filament has run through the gears twice, 1.4mm has gone three times, 1.2mm has gone four times, etc. By the time it finishes that, the last filament has gone through the gears 6 times and is probably a bit chewed up.

Instead, extrude 0.2mm of filament, retract 0.8mm, undo. Repeat 2-3 times before going "now do a perimeter" so that 1-2+ mm of filament is extruded before doing the last 2 extrude/retracts. This way you reduce the number of times the filament will go through the gears and theoretically won't have issues with chewed up filament.

guestisp, 2019-05-10:
Understood.

Jakub-ZiZA, 2020-04-14:
I have similar request (which could help to solve this), but with different intentions.

Situation
Thin wall infill create these small dots (as mentioned above). It is not dots, it is actually very short print moves. These print moves causes:

Addition travel, thus increased print time (main cause I wanted them to remove)
Increased risk of strings
Wear of filament and all other included parts

And usually you don't get stronger printed part, these really short print moves are useless, but you can't disable thin wall infill, because then it will miss in places where it must be. My approach would be a little different, but still so similar that I put it here and not in new issue. Idea:

"Minimal length of extrusion for thin wall infill" or instead of length could be volume (volume is probably better). This would filter out these really short useless print moves, while keep important ones. I see only positives: print time reduced, wear of parts and filament reduced while structural integrity of printed part would not suffer (or at least not so noticeable). It could satisfy original issue creator also.

bsvedin, 2023-03-13:
Is there any progress on this issue?
I would also like this feature

Westicles92, 2023-05-07:
My Orbiter Extruder keeps having issues with grinding the filament when there are many retracts. I use superslicer but being that its based on prusaslicer I would like to see it implemented here.

dontpokejosh, 2024-04-16:
I'm really surprised this doesnt exist in prusa slicer - it exists in CURA and i was tearing my hair out wondering why i was having so much more issues when i printed this fine a year ago with my old slicer :( 

the description by RCmaniac is extremely accurate. turning up distance travelled doesn't help in the situations i am facing without basically turning off retraction all-together

rcmaniac25, 2026-09-13:
Over 7 years later, I still encounter this issue if I left some filament out and happen to have a print with lots of retractions back to back, or dealing with more flexible filaments. So I would still say this is relevant.