prusa3d/PrusaSlicer #2217: slic3r 1.42.0 beta2, z contact distance issue.

wonderland134, 2019-05-11:
### Version
_Version of Slic3r Prusa Edition used goes here_
slic3r PE 1.42.0 beta 2 ver
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
window 10
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
prusa mk3s
### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

When I using support via to slic3r, strange thing was appear. 
If I set the z contact distance to 0.2mm Then I get the actual z distance between last support interface and product bridge to 0.6mm.
I find a mechanism this problem.
If I pick the nozzle size to 0.4mm and line width 0.4mm Then I can get the real z - contact distance = setting value + nozzle size.
So I can reach the minimum real contact z distance near to nozzle size value when I set the z contact distance value to near to 0
Thus, When I use the 0.4 mm nozzle, I can reach closest contact distance approximately 0.4mm.

Below is my wanted.
I want to set actual contact z distance to 0.2mm. But I can't reach this value.
So I think this problem caused unflexibility to set the contact z distance.

I think this setting mechanism will be changed.
Also this problem caused bad quality support contact interface area.

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

faboaic, 2019-05-15:
That issue has been here a few times.
But I really am hoping for a solution since this is the one and only real problem where slic3r prusa edition / PrusaSlicer fails so much compared to other slicers.

With this issue fixed there will be no need for other slicers anymore.

Main problem is that here the bridging is used as first layer on support.

Simplify can do that so much better.

liftbag, 2019-05-15:
If you want z distance to 0.2mm, then enter -0.20mm. Negative values are perfectly considered and calculated.

wonderland134, 2019-05-15:
For liftbag.
I already tried that method.
I hope also try to that way.
Maby result will be amazing!
-0.2 is apply exactly, the way that, insert the support to the product portion.

liftbag, 2019-05-16:
wonderland134, you can verify form the slicing preview. 
The difference between the height of the last dense support and the first layer in contact with it is equal to the size of the nozzle plus the value you enter in the check box. If the value entered is negative, then it will be the size of the nozzle minus the value you enter in the check box.

bubnikv, 2021-03-18:
To improve quality of the object over support, we have added a new settings: "thick bridges". If enabled (that is by default), PrusaSlicer behaves as before. If disabled, bridges are printed as in any other slicer: The extrusion rate is given by the normal layer height x extrusion width * bridge flow ratio. 

[image]

Implemented with ceea9de8b8b1d4b35bdd1180051033680c405ba1 f01f02154c617cc24835475da419ed6322f2ceec

Duplicate of #341 #4126

closed 2021-03-23 by bubnikv