prusa3d/PrusaSlicer #2218: Infil can only be set in 5% increments

Creat, 2019-05-11:
### Version
1.4.2.0-beta2

### Operating system type + version
Windows 10, 1809

### 3D printer brand / version + firmware version (if known)
This is purely a UI bug, but anyway:
Hypercube Evo (default size), Marlin from bugfix-2.0.x branch

### Behavior
It is currently impossible to set infill levels not divisible by 5. Entering a number (with or without the %) doesn't actually get applied. Pressing enter or changing focus both don't work. This applies to both input fields: the one in the side panel of the Plater and the one in print settings under infil.
Selecting a percentage from the drop down works fine, but this only allows 5% increments.

Expected Result: Considering this is a combo-box that allows text input, I'd expect it to accept any number (and it has in the past).

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

ochm, 2019-05-11:
Not, infill can be set by number or by pulldown menu, as you wish, on both places (Plater or Print settings). I see normal behavior.

YuSanka, 2019-05-11:
Yes, this bug is fixed and will be a part of upcoming release. Close as a Duplicate of #2128

closed 2019-05-11 by YuSanka