prusa3d/PrusaSlicer #2219: slice now cause ubantu mate to hang

chimerlock, 2019-05-11:
### Version
_Version of Slic3r Prusa Edition used goes here_
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
     
     Slic3rPE-1.42.0-beta2+linux64-full-201904140843.AppImage

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

     Ububtu 18.04.2 LTS 64 Bit 
     Kernel 4.18.0-18-generic x86_64

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #2168 
     PRUSA MK3S  (not relevent, issue occurs prior to printing)

### Behavior
* _Describe the problem_
       Computer freezes when running slice now. Cannot even open terminal to kill process

* _Steps needed to reproduce the problem_
       load stl file created in FreeCAD
       click slice now

  * _If this is a command-line slicing issue, include the options used_
      MK3S default options for Prusament filement 0.10 mm detail setting 
        Note: after reviewing similar entries, i checked and the default settings have the Avoid crossing perimeters off

* _Expected Results_
     I'm new to Prusa, 3d printing, and slic3r. I oinitially downloaded the beta version because it came up first when the Prusa site linked to github. 
    When slice now repeatedly hung the computer , i downloaded 1.41.final version of prusa slic3r. That version sliced successfully and was able to print, so ill continue to use it. I only opened this issue in case it helps you with the new release. It is possible there is something in the  design of the  stl file that is causing the issue in the new release. the 3mf is attached in a zip file below
     
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_
could not take screen shot due to computer hanging

_Is this a new feature request?_ 
     yes

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

zip file contains the 3mf
[LockGear.zip](https://github.com/prusa3d/Slic3r/files/3168530/LockGear.zip)

mkoistinen, 2019-05-11:
Are you on a multiple display setup? Could it be related to #2220?

chimerlock, 2019-05-11:
I am only using the monitor attached to the laptop, so unlikely related to #2220.

The laptop is an older MacBook pro running Ubuntu mate (it too slow running Mac os so I bought my wife a new one)

Processor is Intel core 2 duo cpu 8700. @ 2.53 GHz X2.

When running the older version with default settings slicer completes in seconds.

chimerlock, 2019-05-22:
Same issue now occurs with new version Slic3r Prusa Edition 2.0.0. Slice now causes system to hang requiring hard reboot

attaching new .3mf causing issue to occur.
[3mffile.gz](https://github.com/prusa3d/PrusaSlicer/files/3209299/3mffile.gz)

additional info:
- drawing created using freecad 0.18.1, mesh created  using standard option  (0.1mm, 30). It also  occurred using mesh created with mefisto (2mm)
-i had successfully slided a couple other smaller shapes in the new version. issue seems to occur only with specific files
-Slice now runs successfully in seconds using older version Slic3r Prusa Edition 1.41.1. 

back to using the old version...

bubnikv, 2019-05-23:
This is unusual as it slices in a snap on my Windows 10 box, and it uses not too much RAM. Are you RAM limited?

chimerlock, 2019-05-23:
The machine has 4gb

chimerlock, 2019-05-27:
I ran previous versions. Previous 1.42.0 successfully slice the file. The issue was introduced with 1.42.0-beta2-, and continues with version 2.2.0. Something changed in this version that causes notebook to hang

bubnikv, 2019-06-03:
A user space application cannot easily make a modern operating system to hang. Usually it is an excess memory allocation or graphics card driver, that may make your laptop to hang. 

You may try https://unix.stackexchange.com/questions/34334/how-to-create-a-user-with-limited-ram-usage

to avoid your machine to hang due to excess memory allocation and swapping.

Compared to the old Slic3r PE, we now allocate a bit more memory due to background processing and due to the higher resolution bed texture.

chimerlock, 2019-06-08:
I am closing this issue. It appears that the issue is related to an apple firmware issue. Another app (FreeCAD) just caused the laptop to hang, so this is definately an issue with the notebook and not an issue with Prusa Slicer. I suspect that the reason that the latest versions of Prusa Slicer triggerred it to hang is related to the memory changes in these releases.

Thanks for your assistance, and sorry for the misleading question

closed 2019-06-08 by chimerlock