prusa3d/PrusaSlicer #2220: Some modal dialogs are opened off-screen and appears to make Slic3r just hang

mkoistinen, 2019-05-11:
### Version
1.42.0 Beta2

### Operating system type + version
MacOS Sierra 10.12.6 (but probably all of them) **with multiple displays**

### 3D printer brand / version + firmware version (if known)
Pruse i3 MK3S + FW 3.7.0

### Behavior
Opening even the Preferences (Configuration > Preferences...) may open the modal dialog off-screen if the user has multiple displays. In my case, I have two displays in this orientation

[image]
_Note, the screen on the left is a "retina" display and actually has more horizontal pixels than the wider-aspect display on the right_

It is my belief that Slic3r is calculating a "midpoint" from _both_ displays from which it displays the modal dialog. In some cases this display area may be off-screen for the screen it is assigned to leaving the user unable to find a _modal_ dialog and I believe many users will just assume that the program has become unresponsive.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
n/a

rtyr, 2021-01-28:
Is this still relevant with PrusaSlicer 2.3.0?

lukasmatena, 2022-04-05:
Hopefully it isn't, nor in 2.4.1. Closing.

closed 2022-04-05 by lukasmatena