prusa3d/PrusaSlicer #2221: More options for the layer points instead of only "filament change"

mkoistinen, 2019-05-11:
### Version
1.42.0 Beta2

### Operating system type + version
MacOS Sierra

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S + FW 3.7.0

### Behavior
I love the way that the layer preview now allows for the addition of filament change points. And, the way it is implemented is much clearer than Prusa Control. _Thank you!_. However, I use these points in the GCode for insertion of magnets or other bits into my prints at specific layers, and while I could certainly continue to edit my gcode as I do now, it would be amazing if when clicking the + icon while in layer preview mode, I could choose from a list of things such as:

1. Filament Change (yes, keep this one);
1. Pause print (ideally, "presenting" the print bed by moving it forward and retracting the print head for easier access to the print);
1. Could be relatively easy to define additional actions with a place in the UI to provide key value pairs with the "Key" being the action name and the value being the GCode to insert.

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
n/a

bubnikv, 2019-08-01:
@YuSanka may look into adding a custom G-code for the manual "color print" event, that would fit your bill. The implementation is a bit more involved now though, as we will support the color change time estimate in the upcoming PrusaSlicer 2.1.0-alpha1.

bubnikv, 2019-08-01:
duplicate of #1899

icanhazpython, 2019-09-01:
I would love to see this as well. I currently do a find/replace in a text editor to replace M600 with the following:
```
G91                  ; Put in relative mode
G1 Z10               ; Lower bed by 10mm
G90                  ; Put back in absolute mode
G1 X0 Y0             ; Zero (home) the X & Y
M300 S300 P1000      ; play beep sound to signal attention
M0 Click To Restart  ; Pause and wait for the user
G91                  ; Put in relative mode
G1 Z-10              ; Raise the bed back up 10mm
G90                  ; Put back in absolute mode
```
The ability to specify this custom code instead of M600 for color change layers would be ideal.

James-G4CLF, 2019-11-16:
I do support Option 2 in mkoistinen's list - with "filament change", in addition to the slight inconvenience of trimming and reinserting the filament the print head position can make inserting the nut/magnet/blade/object to be smuggled rather inconvenient.

James-G4CLF, 2019-11-16:
Remember Murphy's Law*. The problem with manual Gcode edits is the possibility of mistakes. Putting in an "Insert Object" function, once it's debugged, is a really good idea. Using "Change Filament" does not (at present anyway) leave easy access for inserting nut, magnets, blades or things to be smuggled.

* "If it can go wrong, it will go wrong."

McPcholkin, 2020-02-24:
> The ability to specify this custom code instead of M600 for color change layers would be ideal.

Exact! just field in printer custom g-code section,  maybe "color change command" and default it be M600 but if i need use custom g-code (M0 for example) i just change it in printer profile.

neophyl, 2020-02-24:
The 2.2.0 alpha's can do just this already.  Instead of clicking on the + for insert colour change, you can right click it which gives the options Add Colour Change M600, add pause print M601 or Add Custom Gcode which pops up a box allowing you to enter your gcode.

lukasmatena, 2020-02-24:
@neophyl Yes, this issue should be closed. It was already implemented.

closed 2020-02-24 by lukasmatena

McPcholkin, 2020-02-24:
> nstead of clicking on the + for insert colour change, you can right click

I talk about change "+" function to any g-code, i not use M600 at all, but button "+" is useless, so if this button can insert any custom g-code it may be more comfort to use just in one click instead right click - insert custom g-code - ok.

3 steps instead 1 click.