prusa3d/PrusaSlicer #2223: [Feature request] Save project .3mf alongside the .Gcode on edit.

jokohoko, 2019-05-12:
### feature request for upcoming versions

It would be awesome to have a setting to always save the .3mf file alongside the Gcode when saving. This way as prints sometimes fail, or when we want to reprint with improved settings, the raw file is always still available. 
Also now it is hard to see the Gcode to see what was inside, and this way the 3D data is also always there. 
SD card memory so far is more then plenty.

guestisp, 2019-05-12:
You can import a gcode and use the settings stored inside, there is no need to manually see the gcode

jokohoko, 2019-05-12:
But to change settings and re-print them. I always have to save the project or re-load and place all parts.

neophyl, 2019-05-13:
My normal workflow at the moment is to always slice the project files, check that the slice looks good using the preview and once I am happy with it to then save the project as a 3mf file.  Its only after that that I will export the gcode.  
So your request is you want to remove the step of having to save the project file ?  Quite often the name I want for the project and the gcode are different due to the way slic3r names things after the first file added.  Id much prefer for the gcode default file name to be taken from the project name (if saved) as that would be almost as quick and more useful.  At least to my workflow.

cellardoor, 2020-04-13:
I would be very happy 2 use this Feature aswell

closed 2026-09-29 by prusaslicer-bot[bot]