prusa3d/PrusaSlicer #2224: Possible wipe tower improvement

tsamisa, 2019-05-12:
## Version
1.42.0 Beta2

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
mk3s + mmu2s

### Behavior
When i print with mmu2s  sometimes after a couple of layers the wipe tower  start deforming. Especially with PVA soluble material (even with prusa's predefined settings). Lines are popping up. and get dragged by the nozzle. I was wondering instead of having the wipe tower being formed in a linear manner (every new layer just lines above the previous ones) why not having some kind of grid with every layers lines being perpendicular to the previous layer. This will reduce this kind of problems.

_Is this a new feature request?_
Yes

xiphmont, 2019-06-08:
Agreed, and I might need to look into this myself as it's popping up regularly in my own prints now that I'm leaning harder on the MMU2.

m3cinc, 2019-08-06:
An alternative approach would be to provide multiple wipe regions for each filament, to avoid temperature and compatibility issues in deforming the wipe pad and dragging debris all over the build plate and print!

tofb15, 2019-08-10:
Was just about to issue the same thing! A grid would be better for the wipe tower than the current linear approach.

tmcdermaid, 2019-10-07:
I believe the issue I continually experience when using the Prusa i3 MK3S MMU2S to print multi-colors fit’s in this topic. After a few color changes, the wipe tower starts to get very sloppy, ends curl, filament catches the nozzle, etc. as shown in the attached photos. My z-height is good. I’ve adjusted the PrusaSlicer Printer Setting>Extra Loading Distance from the default -25 to -15 but no change. It looks like the wipes do not overlap the external perimeter; basically, the appear to stop short leaving the ends curled which eventually catch the nozzle and begins dragging across the wipe tower. Regardless of the model (i.e., G-Code pre-sliced by the Prusa Team, re-sliced using PrusaSlicer, filament types PLA or PETG), the wipe tower results are the same. Any thoughts?

[image]
[image]
[image]

lukasmatena, 2019-10-07:
@tmcdermaid Is the benchy print ok? It does not look so, but maybe it's just a bad photo.

If it happens with the pre-sliced models and with any material, I'd say it is quite unlikely to be PrusaSlicers fault. Have you properly calibrated your extrusion flow? Does it only affect prints with the wipe tower? Are you printing at the correct temperature?

tmcdermaid, 2019-10-08:
The Benchley print was okay earlier in the print but as the wipe tower starts to come apart, it affects the model being printed. My extrusion flow is good and all prints using a single filament are perfect. However, when I use more than a single color with the wipe tower, it will look like the photos every time regardless of the filament type and regardless of pre-sliced models or models I slice using PrusaSlicer... I’m printing with the Prusa preset temperatures for PLA and PETG. I’ve tried increasing the temp for PETG and decrease from preset to no avail. I’ve tried different brands of filament to include newly unpackaged Prusament but no change when using a wipe tower. I’m truly stumped.

xiphmont, 2019-10-08:
This looks like... you're mixing materials?  PETG and PLA do not adhere.  In fact, one makes a good support material for the other since it will always separate cleanly.  In other words, this looks like a different problem than what is being reported.

tmcdermaid, 2019-10-10:
No, I wish it was as simple as that... All colors in the above examples are PETG. The same results when I Print with PLA.

rtyr, 2021-01-28:
It should be improved in the current version, since the speed on the wipe tower is now limited to the max volumetric speed. Also, looking at the photos it seems like the main issue was not related to PrusaSlicer.

Closing.

closed 2021-01-28 by rtyr