prusa3d/PrusaSlicer #2225: [Feature request] Modify inbuilt start gcode.  (add preheat hot end feature)

Promite2501, 2019-05-13:
Add a preheat to the start gcode.  Curently the bed heats then the the hot end will heat after. Th hot end starts at the same time the ABL starts but the printer will wait after the abl is done so the hot end finishes the preheat.  I notice other silcers will heat both at the same time.  I do prefer the the slic3r method of waiting so the hot end does not over heat but I hate waiting. it would be good if we could add an option to pre heat the hot end a bit. and we can in put the temp.   I would like to start the pre heat hot end about 150 dec c. (not to hot to cook anything if the bed take a really long time)  its a better starting point and will reduce wait times.  This could be added to the Filament Tab with the other temperature settings. 

So at the moment silc3r is exporting this at the start;
M190 S60 ; set bed temperature and wait for it to be reached
M104 S210 ; set temperature
G28 ; Home 
G29 ; BLTOUCH MESH

It would be good to be able to add a pre heat to the start;
M104 S150 ; Pre-Heat Hot End During Bed heating
M190 S60 ; set bed temperature and wait for it to be reached
M104 S210 ; set temperature
G28 ; Home
G29 ; BLTOUCH MESH

Thanks.

### Version
_Slic3r Prusa Edition
Version:   1.42.0-beta1+win64
Build:     Slic3rPE-1.42.0-beta1+win64-full-201903231838

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 17763), 64-bit edition

### 3D printer brand / version + firmware version 
Creality Ender 3 With BL-Touch abl running latest TH3D Firmwear.

ochm, 2019-05-13:
Why not to create your own printer profile with your own Start G-Code?

Promite2501, 2019-05-13:
Yes I have done that, but this code is generated by the slicer.  the G28 on wards is my own code I inserted as you suggested. But the temperature settings are inserted before custom gcode. 

I guess it might work if i set all temps in the slicer to 0 then add custom code as a work around, but I thinking an inbuilt feature would be helpful to others too, as apposed to a work around.

jasontibbitts, 2019-05-16:
This is what I see with current git master:
```
M104 S215 ; set extruder temp
M140 S60 ; set bed temp
Tx
M190 S60 ; wait for bed temp
M109 S215 ; wait for extruder temp
```
That's using the profile for a Prusa Mk3/MMU2S.  But this just comes from the "Start G-code" option, which I haven't changed from the Prusa-provided default:
```
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
Tx
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
```
As far as I can tell, it currently does what you want, and you do indeed have complete control of the gcode which sets the temperatures.

Promite2501, 2019-05-17:
Hi Jasontibbitts,

Thats interesting I just went and did some testing with the latest RC build.
I tested my current profiles I have made for my ender 3, Some on the in built prusa profiles and a new custom profile.  

With the prusa profiles I found its like you have.
```
 M104...
 M140...
 M190...
 M109...
```

But for any of the custom profiles it only out puts gcode with.
```
M190...
M104...
```

It would be awesome if there were options to control this out put.  Ideally my start gcode would look something like this.
```
 M104 S150  ; set low temperature hot end, pre heat
 M190 S60 ; set bed temperature and wait for it to be reached
 M104 S210 ; set temperature hot end, final heat during ABL
 G28 ; Home
 G29 ; BLTOUCH MESH
 ...
```

So I could set all my temps to 0 and add this in custom gcode, but i would be much nicer to have preheat options in the temperature settings.

quarky42, 2019-05-27:
I agree.  I would also like to be able to have control over the Initialization Block and the Pre-Print Block of GCode (AKA Nozzle Ready Block).      Heck, if it were up to me I might even call it the "Pre-Print / Nozzle Purge Block" to indicate that this is the section of code that happens immediately before the model printing starts.

If I could, at a minimum, have access in the GUI to a block that happens AFTER the M109 but before the actual print starts, I could add a special purge that would move the nozzle off of the bed, purge 15mm of filament at a height of about 1 or 2mm off of the bed.  The rapid towards the start of the print would remove the string and the print would start with a freshly primed nozzle. 

I think I can create a version of this in the start g code section if I can add my own M109 like this:
M109 S{first_layer_temperature}  ; wait for extruder temp
followed by my purge commands...

However is that supposed to use {} or [] to get it to actually parse the first_layer_temperature and replace that with an actual value?

bubnikv, 2019-06-03:
> However is that supposed to use {} or [] to get it to actually parse the first_layer_temperature and replace that with an actual value?

https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-Prusa-Edition-Macro-Language

bubnikv, 2019-08-01:
As @jasontibbitts stated, 
> As far as I can tell, it currently does what you want, and you do indeed have complete control of the gcode which sets the temperatures.

Could we close this issue?

Promite2501, 2019-08-05:
Hi bubnikv,

I don't really think this is suitable.  Yes I can edit my start code but I can not edit the first part as it set in the software.  I could set my temps on the filament tab to 0 and then set the required temps in the custom start gcode.  But i feel that this is a work around.   

I do like the macro you linked to.  where I can set ( -5 or -10)
> M190 S{first_layer_bed_temperature[0] - 5}  ; wait for bed temperature - 5
> M140 S[first_layer_bed_temperature]         ; continue bed heating using legacy [<variable>] syntax
> M109 S[first_layer_temperature]             ; wait for nozzle temperature'
but setting this in the custom code is a work around.

my suggestions is something like one of the following:

- Edit the custom printer start temps to be the same as the Prusa printers.

- A option in the filament tab to add a preheat nozzle setting option by adding M104 to the very start of the gcode.  where I could set ~140 dec c..

- Add a setting in the filament/temp tab to do M190 S{first_layer_bed_temperature[0] - 5}

xoniq, 2019-08-16:
I found this issue because I have something similar. I wanted to heat-up the hotend before the custom start gcode was running because I do purge lines, and after the purge line it's gonna wait a bit for the hotend some more.

I just added
```
M109 S[first_layer_temperature] ; set temperature and wait for it to be reached
```
to my start gcode, and now the `M109` added by the software after my custom gcode disappeared. So for me it's solved while starting to reply on this issue.

Also changed this for the bed, I overrule the bed heating in my custom gcode, and PrusaSlicer is smart enough to understand my action and doesn't interfere with it. This is now in the top of my start gcode:

```
M140 S60 ; Pre-set bed temperature, bed heats together with hotend
M109 S190 ; set temperature and wait for it to be reached
M190 S60 ; after hotend is heated, continue to wait for bed temperature to finish
```

I'm gonna test it now for the first time, but in theory, this should give the bed the command to heat up to 60, while also giving the hotend the command to heat up to 190, and then wait for the hotend to heat up, after this, the bed should be still heating, and it will then wait for the bed to finish.