prusa3d/PrusaSlicer #2226: [Question] macOS Build Settings

tcm0116, 2019-05-13:
### Version
Slic3r Prusa Edition 1.42.0-beta2

### Operating system type + version
macOS High Sierra 10.13.6

### Behavior
When following the [Building PrusaSlicer on Mac OS](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Mac%20OS.md) directions, the resulting build crashes in places that the official build for that version does not. For example, the following sequence will result in a Segmentation Fault in my build but not in the official build:

1. Enable SEMM
2. Turn on the Wipe Tower
3. Slice a multi-material part
4. While still in the preview mode, go to the Print Settings and change the Wipe Tower position

Would it be possible to share the details of the system used to build the official builds and if the process is any different than in the instructions?

bubnikv, 2019-05-14:
Thanks for the report. @FidelCapo please test it. The crash may be due to the new feature - rotation of the wipe tower with the rotation gizmo.

> Would it be possible to share the details of the system used to build the official builds and if the process is any different than in the instructions?

We are building slicer with the same build process as we described. If you just replace your slicer binary in the .dmg file, you will get exactly what pops out from our build server.

lukasmatena, 2019-05-24:
I don't think the rotation gizmo is to blame if the crash occurs without it being used. I would bet on some backend-frontend synchronization problem - switching to Preview and changing position of the wipe tower repeatedly in Print Settings behaves weird on my Linux build too - the preview position basically lags one change behind, although I get no crash nor an ASAN alert. I can imagine that whatever causes this could make the application crash if e.g. the order of wxWidgets events was a bit different (which in theory can happen on your setup).

What wxWidgets version you link against? Could you provide a backtrace?

lukasmatena, 2021-01-11:
I assume this is no longer relevant in 2.3.0. The issue that I mentioned should be fixed too. Closing.

closed 2021-01-11 by lukasmatena