prusa3d/PrusaSlicer #2227: Unexpected results with Random seam position

BirkBinnard, 2019-05-13:
[image]
### Version
141.3

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Win10/64

### Behavior
Is this the expected behavior when specifying Random for seam location? Or is there some other parameter I should be looking at?

Or is this fixed in the 142.0 versions?

ochm, 2019-05-13:
What is wrong? Your settings allow stringing. Use google to find how to avoid filament stringing.

BirkBinnard, 2019-05-13:
I know how to avoid stringing. Notice that photo shows the strings happen only on a small portion if the print. Most of it is string-free. What's wrong is that most of the strings start or end at the same place on adjacent layers. This suggests to me that those places are the seams for those particular layers, and they are lined up. Random is supposed to prevent that.

ManuelMcLure, 2019-05-13:
I've found that if you have any case where there's a retraction "random" doesn't really work all that randomly - I was printing gears for a clock and if I printed more than one gear at a time I'd end up with the locations where the travel moves between gears started and ended being all on a single tooth instead of spread out along all the teeth.

BirkBinnard, 2019-05-14:
Manuel - that is very interesting and telling information - thanks for posting it.

What it tells me is that the results I showed will happen when there is a discontinuous path from one part of a given Z-value toolpath to another part at that same Z-value. The toolpath generated for the part I showed apparently has such a discontinuity - perhaps because the end of the spiral is not connected to anything else.

I'm printing another part now (with ver. 1.42.0-beta2) that has no such discontinuity in any of the tool paths, and after about 6 hours of printing there are no strings at all. So perhaps whatever caused the problem with ver. 141.3 is now fixed.

haarp, 2019-07-21:
I can confirm that random seams are not random. Prusaslicer has a strong tendency to align seams regardless. Attached is a screenshot showing such alignment with seams set to "random".

[image]

Retraction doesn't appear to make a difference, the seams are still aligning with retraction disabled.

rtyr, 2021-01-28:
Should be fixed in 2.3.0. Closing.

closed 2021-01-28 by rtyr